ai-winter/ros_motion_planning #3: Can I add multiple robots in this program?

lgl-960709, 2023-03-03:
Can I add multiple robots in this program? How do I add them. Thank you very much for your help!

ai-winter, 2023-03-03:
Hello, @lgl-960709 

Yes, the environment supports multi-agent simulation. Please refer to the `README.md` file, `robot_i` stands for the i-th agent in the environment.

```
map: "warehouse"
world: "warehouse"
robots_config:
  - robot1_type: "turtlebot3_burger"
    robot1_global_planner: "astar"
    robot1_local_planner: "dwa"
    robot1_x_pos: "0.0"
    robot1_y_pos: "0.0"
    robot1_z_pos: "0.0"
    robot1_yaw: "-1.57"
  - robot2_type: "turtlebot3_burger"
    robot2_global_planner: "jps"
    robot2_local_planner: "pid"
    robot2_x_pos: "-5.0"
    robot2_y_pos: "-7.5"
    robot2_z_pos: "0.0"
    robot2_yaw: "0.0"
robots_init: "robots_rviz_init.yaml"
rviz_file: "sim_env.rviz"
```

lgl-960709, 2023-03-03:
Thank you very much for your generosity, thank you for your help, I wish you a happy life!

lgl-960709, 2023-03-03:
Sorry to bother you again, I'm a ROS beginner. I copy RAMD.me in user_ After config.yml, there are no two cars displayed. What else do I need to change. I would be very grateful!

ai-winter, 2023-03-04:
Hello, @lgl-960709 

Please set `rviz_file` as null, i.e. `rviz_file: ""`, and try again

closed 2024-01-19 by ZhanyuGuo