ai-winter/ros_motion_planning #6: catkin_make did not pass

humble0628, 2023-03-18:
strian_sfm_plugin/src/pedestrian_sfm_plugin.cpp:133:25: error: ‘ignition::math::v4::Pose3d {aka class ignition::math::v4::Pose3<double>}’ has no member named ‘X’
       goal.center.set(g.X(), g.Y());

errors are above. i have search via google but found nothing

ai-winter, 2023-03-18:
Hello, @humble0628 

Could you please provide us with more information about your machine? Such as version of `Ubuntu`, `Gazebo`, etc.

humble0628, 2023-03-18:
> Hello, @humble0628
> 
> Could you please provide us with more information about your machine? Such as version of `Ubuntu`, `Gazebo`, etc.

Thanks for your reply!
I use Ubuntu18.04 and gazebo9.0.

ZhanyuGuo, 2023-03-18:
Hello, @humble0628, and thanks for your supporting!

We tested our code on `ubuntu18` and encountered the same error, and we noticed that `ignition-math4` will be used on `ubuntu18` instead of `ignition-math6` on `ubuntu20`. 

In `ignition-math4`, `Pose3d` does not have the member `X()` or `Y()`, and instead we need to get them through `Pose3d::Pos()::X()`. We have fixed it in the commit af03710.

Thank you for finding this bug, this issue will be closed.

closed 2023-03-18 by ZhanyuGuo