ai-winter/ros_motion_planning #7: l stop here and can not open gazeboll .it is dark

tl123987, 2023-03-28:
[INFO] [1680004959.181575, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1680004959.185961, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1680004959.307924071]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1680004959.309054772]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1680004959.353289594]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1680004959.354068298]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1680004960.756150750]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1680004960.768146522]: Physics dynamic reconfigure ready.
[INFO] [1680004960.995906, 0.000000]: Calling service /gazebo/spawn_urdf_model

> 
[image]

tl123987, 2023-03-28:
[image]

ai-winter, 2023-03-28:
Hello, @tl123987 

Do you try to move these model files to the `.gazebo/models/` directory?

tl123987, 2023-03-28:
I think maybe the plugs caues this

ai-winter, 2023-03-28:
@tl123987 

Recently, we updated the pedestrians Gazebo plugin to construct multi-agent environment. However, we found that this update had poor compatibility. In order for us to locate issues and improve the project, could you please provide us with information about your system environment? Or if you have found the bugs, welcome to share your experience with us. Thank you!

tl123987, 2023-03-29:
[image]

I removed the latter stuff，now   l open it but can see the world

tl123987, 2023-03-29:
[image]

Why is there always a bunch of things behind the spawn

ai-winter, 2023-03-29:
Hello, @tl123987 
We designed a script to load pedestrian plugin automatically, you can disable it by setting `pedestrian: ""` in `user_config.yaml`

tl123987, 2023-03-29:
This plugin doesn't work, but your other plugin project does

ZhanyuGuo, 2023-03-29:
Hello @tl123987, and thanks for your supporting!

We are sorry to solove this problem so late...

We have tested our code on an entirely pure ubuntu 20.04 with ROS-noetic, following

1. Install [ROS](http://wiki.ros.org/cn/noetic/Installation/Ubuntu), Desktop-Full.

2. Install git: `sudo apt install git`.

3. Clone the reposity: `git clone https://github.com/ai-winter/ros_motion_planning.git`.

4. Other dependence.

    ```bash
    sudo apt install ros-noetic-amcl ros-noetic-base-local-planner ros-noetic-map-server ros-noetic-move-base ros-noetic-navfn python-is-python3
    ```

5. Move or copy files in `src/sim_env/models/` into `~/.gazebo/models/`.

6. Run the scripts in `src/sim_env/scripts/`, i.e. `cd src/sim_env/scripts/ && ./main.sh`. 

    **NOTE: Changing launch files DOES NOT work, because some of them are regenerated according to the `user_config.yaml` by our python script when you run `main.sh`. Therefore, you should change configs in `user_config.yaml` instead of launch files.**

We found that the models in this reposity are not accurate, and we checked them. We have added and deleted some models in 6b16d8d. You can now pull the latest code in master branch and test it.

Hope there will be no problem. 🙏

Sorry again for our carelessness...

ZhanyuGuo, 2023-03-29:
> Hello @tl123987, and thanks for your supporting!
> 
> We are sorry to solove this problem so late...
> 
> We have tested our code on an entirely pure ubuntu 20.04 with ROS-noetic, following
> 
> 1. Install [ROS](http://wiki.ros.org/cn/noetic/Installation/Ubuntu), Desktop-Full.
> 2. Install git: `sudo apt install git`.
> 3. Clone the reposity: `git clone https://github.com/ai-winter/ros_motion_planning.git`.
> 4. Other dependence.
>    ```shell
>    sudo apt install ros-noetic-amcl ros-noetic-base-local-planner ros-noetic-map-server ros-noetic-move-base ros-noetic-navfn python-is-python3
>    ```
> 5. Move or copy files in `src/sim_env/models/` into `~/.gazebo/models/`.
> 6. Run the scripts in `src/sim_env/scripts/`, i.e. `cd src/sim_env/scripts/ && ./main.sh`.
>    **NOTE: Changing launch files DOES NOT work, because some of them are regenerated according to the `user_config.yaml` by our python script when you run `main.sh`. Therefore, you should change configs in `user_config.yaml` instead of launch files.**
> 
> We found that the models in this reposity are not accurate, and we checked them. We have added and deleted some models in [6b16d8d](https://github.com/ai-winter/ros_motion_planning/commit/6b16d8d0be20a52730c61dfdaa5a0ea44e49012e). You can now pull the latest code in master branch and test it.
> 
> Hope there will be no problem. 🙏
> 
> Sorry again for our carelessness...

In short, @tl123987 you may just need to pull the latest code and copy the files in `src/sim_env/models/` into `~/.gazebo/models/`.

We will improve our readme file about the dependence and how to run.

tl123987, 2023-03-31:
thanks for your answer.NOW l try it ,it is no problem.thanks very much

closed 2023-03-31 by ZhanyuGuo