ai-winter/ros_motion_planning #8: Hello, thank you for sharing! May I ask how to disable the global path planning algorithm? I would like to use only the local path planning DWA algorithm for experimentation.

chengrj, 2023-04-07:
Hello, thank you for sharing! May I ask how to disable the global path planning algorithm? I would like to use only the local path planning DWA algorithm for experimentation.

ai-winter, 2023-04-07:
Hello, @chengrj 

Now there is no way to disable the global path planning. We are sorry about it. However, we plan to implement a planner that publishes empty plans to meet your requirement in the future.  Welcome your continued interest in the project.

chengrj, 2023-04-07:
May I ask if it is enough to leave the option 'robot1_global_planner' blank in the user_config.yaml file?

chengrj, 2023-04-07:
May I ask which global path planning algorithm is combined with dwa.gif in readme.md to achieve the desired effect?

ai-winter, 2023-04-07:
Hello, @chengrj

Setting `robot1_global_planner: ` is equivalent to disabling the global planner plugin, which will enable the default planner called `Navfn` in ROS, namely there is also a global path during navigation.

chengrj, 2023-04-07:
Thank you for your answer! May I replace it with my own map file? How to achieve it?

ai-winter, 2023-04-07:
Sure, @chengrj. You can achieve it by replacing `map` item in `user_config.yaml` with your own map.

closed 2023-04-09 by ZhanyuGuo