ai-winter/ros_motion_planning #9: Step 5: Question reagrding '~/.gazebo/models/'

Ashwini-Rathi, 2023-04-14:
Hi,
I am a beginner and unable to find '~/.gazebo/models/'. Where can I find this location?

ai-winter, 2023-04-14:
Hello, @Ashwini-Rathi 

`.gazebo` is a hidden directory, you can find this by following the steps below：
- `cd ~`
- `Ctrl + H` can be useful to see hidden directories
- Select the `.gazebo`

Hope this could be helpful for you.

Ashwini-Rathi, 2023-04-14:
Hi,
thank yo for the quick reply. I ge the flowing error for the step 6 execution `catkin_make`

```
ash@ubun:~/catkin_ws/ros_motion_planning$ catkin_make
Base path: /home/ash/catkin_ws/ros_motion_planning
Source space: /home/ash/catkin_ws/ros_motion_planning/src
Build space: /home/ash/catkin_ws/ros_motion_planning/build
Devel space: /home/ash/catkin_ws/ros_motion_planning/devel
Install space: /home/ash/catkin_ws/ros_motion_planning/install
Creating symlink "/home/ash/catkin_ws/ros_motion_planning/src/CMakeLists.txt" pointing to "/opt/ros/noetic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/ash/catkin_ws/ros_motion_planning/src -DCATKIN_DEVEL_PREFIX=/home/ash/catkin_ws/ros_motion_planning/devel -DCMAKE_INSTALL_PREFIX=/home/ash/catkin_ws/ros_motion_planning/install -G Unix Makefiles" in "/home/ash/catkin_ws/ros_motion_planning/build"
####
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/ash/catkin_ws/ros_motion_planning/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ash/catkin_ws/ros_motion_planning/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
CMake Deprecation Warning at /usr/src/googletest/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googlemock/CMakeLists.txt:45 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googletest/CMakeLists.txt:56 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 16 packages in topological order:
-- ~~  - dynamic_rviz_config
[... 630 lines cut ...]
[ 64%] Generating Javascript code from pedsim_msgs/TrackedPersons.msg
[ 64%] Generating Python msg __init__.py for pedsim_msgs
[ 64%] Generating Javascript code from pedsim_msgs/TrackedGroup.msg
[ 64%] Generating Python from MSG spencer_tracking_msgs/TrackingTimingMetrics
[ 64%] Generating Python code from SRV spencer_tracking_msgs/GetPersonTrajectories
[ 65%] Generating Javascript code from pedsim_msgs/TrackedGroups.msg
[ 67%] Generating Javascript code from pedsim_msgs/SocialRelation.msg
[ 67%] Generating Javascript code from pedsim_msgs/SocialRelations.msg
[ 67%] Generating Javascript code from pedsim_msgs/SocialActivity.msg
[ 67%] Generating Javascript code from pedsim_msgs/SocialActivities.msg
[ 68%] Generating Javascript code from pedsim_msgs/Waypoint.msg
[ 68%] Built target pedsim_msgs_generate_messages_eus
[ 68%] Generating Javascript code from pedsim_msgs/Waypoints.msg
[ 68%] Generating Python msg __init__.py for spencer_social_relation_msgs
[ 68%] Built target pedsim_msgs_generate_messages_nodejs
[ 68%] Built target pedsim_msgs_generate_messages_py
[ 69%] Generating Python srv __init__.py for spencer_tracking_msgs
[ 69%] Generating Python msg __init__.py for spencer_tracking_msgs
[ 69%] Built target spencer_social_relation_msgs_generate_messages_py
[ 69%] Built target spencer_tracking_msgs_generate_messages_py
[ 69%] Linking CXX shared library /home/ash/catkin_ws/ros_motion_planning/devel/lib/libpid_planner.so
[ 69%] Built target pid_planner
make: *** [Makefile:146: all] Error 2
Invoking "make -j12 -l12" failed
```

ZhanyuGuo, 2023-04-14:
Hello @Ashwini-Rathi. There are still some problems in compiling, therefore, you should follow the stages in README.md.

```bash
cd ros_motion_planning/
catkin_make -DCATKIN_WHITELIST_PACKAGES="gazebo_sfm_plugin;pedsim_msgs"
catkin_make -DCATKIN_WHITELIST_PACKAGES=""

# Afterwards, everytime you just need to
catkin_make
```

We are sorry that we forgot to uncomment them...

Hope it works.

Ashwini-Rathi, 2023-04-14:
Hi,
When I run the command `catkin_make -DCATKIN_WHITELIST_PACKAGES="gazebo_sfm_plugin;pedsim_msgs"`, I get the followign error. can you help me resolve this?
```
[ 98%] Building CXX object third_party/gazebo_plugins/pedestrian_sfm_plugin/CMakeFiles/PedestrianSFMPlugin.dir/src/pedestrian_sfm_plugin.cpp.o
[ 98%] Built target gazebo_sfm_plugin_generate_messages
In file included from /usr/include/ignition/math6/gz/math/Quaternion.hh:20,
                 from /usr/include/ignition/math6/gz/math/Pose3.hh:20,
                 from /usr/include/ignition/math6/ignition/math/Pose3.hh:18,
                 from /usr/include/sdformat-9.8/sdf/Actor.hh:23,
                 from /usr/include/sdformat-9.8/sdf/sdf.hh:2,
                 from /usr/include/gazebo-11/gazebo/common/Plugin.hh:33,
                 from /home/ash/catkin_ws/ros_motion_planning/src/third_party/gazebo_plugins/pedestrian_sfm_plugin/include/pedestrian_sfm_plugin.h:27,
                 from /home/ash/catkin_ws/ros_motion_planning/src/third_party/gazebo_plugins/pedestrian_sfm_plugin/src/pedestrian_sfm_plugin.cpp:18:
/usr/include/ignition/math6/gz/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::duration ignition::math::v6::stringToDuration(const string&)’:
/usr/include/ignition/math6/gz/math/Helpers.hh:1059:28: error: ‘chrono_literals’ is not a namespace-name
 1059 |       using namespace std::chrono_literals;
      |                            ^~~~~~~~~~~~~~~
/usr/include/ignition/math6/gz/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::time_point ignition::math::v6::stringToTimePoint(const string&)’:
/usr/include/ignition/math6/gz/math/Helpers.hh:1101:28: error: ‘chrono_literals’ is not a namespace-name
 1101 |       using namespace std::chrono_literals;
      |                            ^~~~~~~~~~~~~~~
/usr/include/ignition/math6/gz/math/Helpers.hh:1102:56: error: unable to find numeric literal operator ‘operator""s’
 1102 |       std::chrono::steady_clock::time_point timePoint{-1s};
      |                                                        ^~
/usr/include/ignition/math6/gz/math/Helpers.hh:1102:56: note: use ‘-fext-numeric-literals’ to enable more built-in suffixes
/usr/include/ignition/math6/gz/math/Helpers.hh:1102:58: error: no matching function for call to ‘std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >::time_point(<brace-enclosed initializer list>)’
 1102 |       std::chrono::steady_clock::time_point timePoint{-1s};
      |                                                          ^
In file included from /usr/include/c++/9/mutex:39,
                 from /usr/include/boost/system/detail/std_interoperability.hpp:13,
                 from /usr/include/boost/system/error_code.hpp:959,
                 from /usr/include/boost/system/system_error.hpp:11,
                 from /usr/include/boost/thread/exceptions.hpp:22,
                 from /usr/include/boost/thread/pthread/mutex.hpp:14,
                 from /usr/include/boost/thread/mutex.hpp:16,
                 from /opt/ros/noetic/include/ros/publisher.h:36,
                 from /opt/ros/noetic/include/ros/node_handle.h:32,
                 from /opt/ros/noetic/include/ros/ros.h:45,
                 from /home/ash/catkin_ws/ros_motion_planning/src/third_party/gazebo_plugins/pedestrian_sfm_plugin/include/pedestrian_sfm_plugin.h:21,
                 from /home/ash/catkin_ws/ros_motion_planning/src/third_party/gazebo_plugins/pedestrian_sfm_plugin/src/pedestrian_sfm_plugin.cpp:18:
/usr/include/c++/9/chrono:643:14: note: candidate: ‘template<class _Dur2, class> constexpr std::chrono::time_point<_Clock, _Dur>::time_point(const std::chrono::time_point<_Clock, _Dur2>&)’
  643 |    constexpr time_point(const time_point<clock, _Dur2>& __t)
      |              ^~~~~~~~~~
/usr/include/c++/9/chrono:643:14: note:   template argument deduction/substitution failed:
/usr/include/c++/9/chrono:636:21: note: candidate: ‘constexpr std::chrono::time_point<_Clock, _Dur>::time_point(const duration&) [with _Clock = std::chrono::_V2::steady_clock; _Dur = std::chrono::duration<long int, std::ratio<1, 1000000000> >; std::chrono::time_point<_Clock, _Dur>::duration = std::chrono::duration<long int, std::ratio<1, 1000000000> >]’
  636 |  constexpr explicit time_point(const duration& __dur)
      |                     ^~~~~~~~~~
/usr/include/c++/9/chrono:636:21: note:   conversion of argument 1 would be ill-formed:
/usr/include/c++/9/chrono:633:12: note: candidate: ‘constexpr std::chrono::time_point<_Clock, _Dur>::time_point() [with _Clock = std::chrono::_V2::steady_clock; _Dur = std::chrono::duration<long int, std::ratio<1, 1000000000> >]’
  633 |  constexpr time_point() : __d(duration::zero())
      |            ^~~~~~~~~~
/usr/include/c++/9/chrono:633:12: note:   candidate expects 0 arguments, 1 provided
/usr/include/c++/9/chrono:626:14: note: candidate: ‘constexpr std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >::time_point(const std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >&)’
  626 |       struct time_point
      |              ^~~~~~~~~~
/usr/include/c++/9/chrono:626:14: note:   conversion of argument 1 would be ill-formed:
[... cut ...]
/usr/include/boost/array.hpp:446:13: note:   template argument deduction/substitution failed:
make[2]: *** [third_party/gazebo_plugins/pedestrian_sfm_plugin/CMakeFiles/PedestrianSFMPlugin.dir/build.make:76: third_party/gazebo_plugins/pedestrian_sfm_plugin/CMakeFiles/PedestrianSFMPlugin.dir/src/pedestrian_sfm_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3566: third_party/gazebo_plugins/pedestrian_sfm_plugin/CMakeFiles/PedestrianSFMPlugin.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
Invoking "make -j12 -l12" failed
```

ZhanyuGuo, 2023-04-14:
@Ashwini-Rathi, what is your ubuntu or ROS version?

ZhanyuGuo, 2023-04-14:
@Ashwini-Rathi, try change line 4 in `src/third_party/gazebo_plugins/pedestrian_sfm_plugin/CMakeLists.txt` from `add_compile_options(-std=c++11)` to `add_compile_options(-std=c++17)`.

Ashwini-Rathi, 2023-04-14:
ubuntu 20.04 LTS with ROS Noetic.
Changing the version from c++11 to c++17 resolved the issue

ZhanyuGuo, 2023-04-14:
Great!

closed 2023-04-14 by Ashwini-Rathi

ZhanyuGuo, 2023-04-14:
> Hello @Ashwini-Rathi. There are still some problems in compiling, therefore, you should follow the stages in README.md.
> 
> ```shell
> cd ros_motion_planning/
> catkin_make -DCATKIN_WHITELIST_PACKAGES="gazebo_sfm_plugin;pedsim_msgs"
> catkin_make -DCATKIN_WHITELIST_PACKAGES=""
> 
> # Afterwards, everytime you just need to
> catkin_make
> ```
> 
> We are sorry that we forgot to uncomment them...
> 
> Hope it works.

@Ashwini-Rathi, @ai-winter, we do not need this now!

please refer to 24268cf and [message-headers-wont-build-first](https://answers.ros.org/question/188982/message-headers-wont-build-first/)