ai-winter/ros_motion_planning #11: Map not recieved

GeorgeBethel, 2023-05-05:
I get the error below on the terminal

[ WARN] [1683284527.630343657, 90.141000000]: No laser scan received (and thus no pose updates have been published) for 90.141000 seconds.  Verify that data is being published on the /scan topic.
[ WARN] [1683284531.473494152, 91.721000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.
[ WARN] [1683284544.047433434, 96.821000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.

I did not find anywhere you broadcast transform

ai-winter, 2023-05-05:
Hello, @GeorgeBethel 

How did you start the environment? Checking the ROS nodes would be helpful, especially ensure that `AMCL` and `Move Base` node have been successfully executed.

GeorgeBethel, 2023-05-05:
Hello,

Thank you for your response. I started the main.sh file. Move base and amcl
nodes are also running.

GeorgeBethel, 2023-05-05:
Rviz view

[image]

ai-winter, 2023-05-05:
@GeorgeBethel You can try to check out the tf-tree by `rqt_tf_tree` tool. The following image shows the tf-tree running normally, provided for your reference.

[image]

GeorgeBethel, 2023-05-05:
Thank. It has been resolved. The issues was due to some incomplete dependencies. I uninstalled and reinstalled ROS noetic

closed 2023-05-05 by GeorgeBethel

WeiJunREN, 2024-06-21:
> Thank. It has been resolved. The issues was due to some incomplete dependencies. I uninstalled and reinstalled ROS noetic

Hello,could you tell me what dependencies did you reinstalled ? I meet a same question , hh