ai-winter/ros_motion_planning #12: I have a question

yjydiancom, 2023-05-18:
[image]
Which files should I run to get the following results?

ZhanyuGuo, 2023-05-18:
Hello @yjydiancom, and thanks for your supporting! For this issue, you should refer to `README.md`.

To get different results, you need to change the configuration in the `./src/user_config/user_config.yaml`, and options can be seen in about [line 30](https://github.com/ai-winter/ros_motion_planning/blob/9d876149b42b6ab18ef7eb8f89642c91582a6746/src/sim_env/launch/include/navigation/move_base.launch.xml#L30) of [move_base.launch.xml](https://github.com/ai-winter/ros_motion_planning/blob/9d876149b42b6ab18ef7eb8f89642c91582a6746/src/sim_env/launch/include/navigation/move_base.launch.xml).

Hope this helps.

yjydiancom, 2023-05-22:
Okay, thank you very much. I'm learning to understand the code.

closed 2023-05-22 by ai-winter