ai-winter/ros_motion_planning #15: ubuntu 18.04

summer-rise, 2023-05-31:
### ubuntu 18.04 

When I compile by catkin_make, it prompts an error.
`Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Could NOT find utils (missing: utils_DIR)
-- Could not find the required component 'utils'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "utils" with any of
  the following names:

    utilsConfig.cmake
    utils-config.cmake

  Add the installation prefix of "utils" to CMAKE_PREFIX_PATH or set
  "utils_DIR" to a directory containing one of the above files.  If "utils"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  planner/global_planner/graph_planner/CMakeLists.txt:4 `(find_package)`
`
[image]

ai-winter, 2023-06-01:
Hello, @summer-rise 

We fix the bug by updating the dependencies in `package.xml`, you could try again and give us feedback.

summer-rise, 2023-06-02:
> Hello, @summer-rise
> 
> We fix the bug by updating the dependencies in , you could try again and give us feedback.`package.xml`

The same issue again when i redownload and compile

ZhanyuGuo, 2023-06-02:
> When I download and compile with ubuntu 20.04 in VMware, it works successfully. But in a dual system, using ubuntu 18.04, it reported an error, and the problem was the picture I started posting. It says I seem to be missing a package called utils, but I can't install it in ubuntu 18.04.

`utils` is our package, we will keep trying to find the problem.

ZhanyuGuo, 2023-06-02:
Hello @summer-rise.

We have tried build our current reposity in totally new Docker and VMWare of `ros-melodic`, but did not meet your issue... 

We still don't have a clue about this issue. :sob:

summer-rise, 2023-06-02:
> 你好 .
> 
> 我们已经尝试在全新的 Docker 和 VMWare 中构建我们当前的存储库，但没有满足您的问题......`ros-melodic`
> 
> 我们仍然对这个问题一无所知。sob

It's okay, I can build this project in VMWare.

closed 2023-06-10 by ZhanyuGuo