ai-winter/ros_motion_planning #16: Run Error  (D*，D*_lite )

etefron, 2023-06-08:
### Operating environment
Ubuntu 18.06
ROS (Melodic)

### Problem description
In the real machine test, when planner is used as the plug-in of move_base, the compilation is successful, and other global path planning algorithms run without exception. However, when d_star and d_star_lite are used as global path algorithms, the following two error messages (basically the first kind) appear when the path planning algorithm is started at the map boundary:

**The first error message**
[move_base-17] process has died [pid 5793, exit code -11

**The second error message**
terminate called after throwing an instance of 'std::bad_alloc'
what():  std::bad_alloc
[move_base-17] process has died [pid 8520, exit code -6,

Can you look at the reason for the error? Thank you very much.

ZhanyuGuo, 2023-06-09:
Hello @etefron, and thanks for your supporting!

You mentioned that `when the path planning algorithm is started at the map boundary`. I think there may be some bugs when the map is not available. 

We will reproduce and solve your issue as soon as possible.

ZhanyuGuo, 2023-06-11:
Hello @etefron.

We have reproduced and solved your issue in 7004540.

We made a stupid mistake on checking the boundary of the array :joy:.

We have fixed all of them, e.g. line [115](https://github.com/ai-winter/ros_motion_planning/blob/7004540a48ddeb1254a1403a9579159e5c948616/src/planner/global_planner/graph_planner/src/d_star.cpp#L115) in `d_star.cpp`.

etefron, 2023-06-13:
Hello @ZhanyuGuo 
After updating the code, the D* and D* Lite algorithms are now working properly. Thank you for your support and response, and we look forward to your follow-up work.

closed 2023-06-13 by etefron