ai-winter/ros_motion_planning #17: Failed to get a path.

aamir-Asophilia, 2023-07-17:
This is a great project you have here. I tried playing it myself, but I found that the path planning often fails when the endpoint is near obstacles. What could be the reason for this? Best wishes!

ZhanyuGuo, 2023-07-19:
Hello @aamir-Asophilia, and thanks for your supporting!

Based on the robot radius, we inflat the obstacles and plan a path on such a cost map. Therefore, if the end-point is too close to the obstacles, it may actually locate in the inflated obstacles, and it is impossible to find a feasible path.

I wonder if that explains your issue.

aamir-Asophilia, 2023-07-19:
Hello @ZhanyuGuo ,Thank you very much for your answer. However, in practice, there is another situation where it occurs. If a path has already been planned and the machine is walking along it, it may stop at a certain point if the path is too close to an obstacle. Is this also the same problem?

ZhanyuGuo, 2023-07-19:
Hello @aamir-Asophilia. If the path disappears in rviz, I think it is the same problem: the car is too close to the obstacle and it is considered to be in the inflated obstacle.

This is a troubling problem that we have noticed during development.

The planned path is always close to the inflated obstacle, and the control may be overshoot. Therefore, the car will sometimes locate in the inflated obstacle, and it is impossible to find a feasible path when the start-point is in the obstacle.

We reduce the gain of control to reduce the overshoot. And we pray that when the path is lost, the car can track the previous one to return to the normal area.

Do you have any good ideas? Thank you very much!

aamir-Asophilia, 2023-07-19:
Yes, this is a bit troublesome. It may require matching different algorithms based on the usage scenario. This may involve business logic problems, making it difficult to have a unified method to handle it. However, thank you very much for your answer, thank you.

ZhanyuGuo, 2024-01-19:
This issue is still unsolved and worth studying.

ZhanyuGuo, 2024-01-20:
This issue may be solved by this [line](https://github.com/ai-winter/ros_motion_planning/blob/6881eac50b9a1af3d395be8b3e52d6b5ddced8b1/src/core/global_planner/graph_planner/src/a_star.cpp#L112).

That is, our definition of an obstacle point on the costmap requires that its cost not only exceed a threshold, but also exceed the point where the robot is currently located.

closed 2024-01-20 by ZhanyuGuo