ai-winter/ros_motion_planning #22: source directory in third party directory missing ped_state.h

kashifzr, 2023-07-31:
Errors     << gazebo_ped_visualizer_plugin:make /home/rob-kkj/home/ros_motion_planning/logs/gazebo_ped_visualizer_plugin/build.make.000.log
In file included from /home/rob-kkj/home/ros_motion_planning/src/ros_motion_planning/src/third_party/gazebo_plugins/pedestrian_visualizer_plugin/gazebo_ped_visualizer_plugin/src/pedestrian_visual_plugin.cpp:18:
/home/rob-kkj/home/ros_motion_planning/src/ros_motion_planning/src/third_party/gazebo_plugins/pedestrian_visualizer_plugin/gazebo_ped_visualizer_plugin/include/pedestrian_visual_plugin.h:38:10: fatal error: gazebo_sfm_plugin/ped_state.h: No such file or directory
   38 | #include <gazebo_sfm_plugin/ped_state.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/PedestrianVisualPlugin.dir/build.make:76: CMakeFiles/PedestrianVisualPlugin.dir/src/pedestrian_visual_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2451: CMakeFiles/PedestrianVisualPlugin.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
cd /home/rob-kkj/home/ros_motion_plann

ZhanyuGuo, 2023-08-01:
Hello @kashifzr, and thanks for your supporting!

try as follows:
```
catkin_make -DCATKIN_WHITELIST_PACKAGES="gazebo_sfm_plugin;pedsim_msgs"
catkin_make -DCATKIN_WHITELIST_PACKAGES=""
```
We have fixed the issue in a previous version, but I don't know if you have updated to the latest version. If you have updated and still meet this issue, please tell us more about your system version information and we will try reproducing it.

kashifzr, 2023-08-01:
thanks Now its works

closed 2023-08-07 by ai-winter