ai-winter/ros_motion_planning #23: How to set the resolution?

aamir-Asophilia, 2023-08-07:
Hello, I'm here to help. In the process of using it, how can I modify the resolution? How can I match the algorithm according to the actual size of the machine? 
Looking forward to your response. Thank you!

ai-winter, 2023-08-07:
Hello, @aamir-Asophilia 

We are sorry, but this project currently does not support robot resolution adjustment. You can create a robot model that matches the physical dimensions using SolidWorks and then import that model into the platform for use.

aamir-Asophilia, 2023-08-07:
Thank you very much for your response! My apologies for the lack of accuracy in my previous statement. What I meant was, if my resolution is 0.05, do I need to modify the grip2map function and other related grid map functions? Additionally, if my model has a radius of 0.5m, which parameters should be modified in the cost map?

I am currently facing an issue where, when using the grap_planner plugin without modifying any code, the resolution of the generated paths doesn't seem to be 0.05. I'm unsure of what could be affecting the resolution of the search paths. I feel quite frustrated, so I would greatly appreciate any guidance or suggestions you can provide. Thank you very much!

ai-winter, 2023-08-07:
You can try modifying the `resolution` item in the file `map.yaml`. Functions like `grid2map` actually read the resolution from this file, which determines the size of the grid map.

closed 2023-12-21 by ai-winter