ai-winter/ros_motion_planning #24: can't find file

ambglh, 2023-08-09:
:~/ros_study/src/ros_motion_planning/src/sim_env/scripts$ ./main.sh 
RLException: [main.launch] is neither a launch file in package [sim_env] nor is [sim_env] a launch file name
The traceback for the exception was written to the log file
how can I solve this problem?

ambglh, 2023-08-09:
I have used catkin_make
ghm@ghm:~/ros_study/src/ros_motion_planning/src/sim_env/scripts$ rospack find sim_env 
/home/ghm/ros_study/src/ros_motion_planning/src/sim_env

zhangxu140823, 2023-08-09:
Thank you for your program!   I want to solve the problem as follow:
/ros_motion_planning/src/third_party/gazebo_plugins/pedestrian_visualizer_plugin/gazebo_ped_visualizer_plugin/include/pedestrian_visual_plugin.h:38:10: fatal error: gazebo_sfm_plugin/ped_state.h:  
not found ped_state.h

ZhanyuGuo, 2023-09-05:
Hello @zhangxu140823, please refer to #22.

ZhanyuGuo, 2023-09-05:
Hello @ambglh. We have tried it in a clean environment, but still not encountered this issue.

I notice that you put the code in your own ROS workspace. I think you can try moving the reposity to a new place, not sourcing other workspace (maybe in your `~/.bashrc`, e.g. `source xxx/devel/setup.bash`, comment it and open a new terminal), and following our readme again.

closed 2023-12-21 by ai-winter