ai-winter/ros_motion_planning #28: Questions about the controller

2892510130, 2023-08-29:
First of all, I would like to express my appreciation for this project, it's really helpful for me to understand how to write a planner by myself! 

Robot navigation can be divided into several steps: mapping, localization, global plan, local plan and control. I want to know if I can control a mobile robot well in the real world using only a DWA planner or PID planner(it is not a controller, right? haven't read it.), or if I need to write a controller. 

In addition, whether I need to write a controller or not, I would like to know how to write a controller myself. Is there a resource for this topic(control), preferably one as clear as yours(planning)? I am looking forward to your reply!

ai-winter, 2023-08-30:
Hello, @2892510130 .

Thank you for your affirmation of the project.

The local planner and controller have fundamentally different roles. The goal of the local planner is to determine the current velocity $v$ based on the environment and the global path. On the other hand, the controller's objective is to drive the robot's motors to achieve this desired velocity. Therefore, PID has different meanings in the planner and the controller.

In real-world, robots typically come with built-in controllers that can be loaded through launch files. In this case, our focus is primarily on generating reasonable velocity profiles through planner. However, if you wish to improve the control performance, further development of controller code may be necessary.

2892510130, 2023-08-30:
Thank you for your reply! Are there any resources that can teach me how to write my own controllers?

ZhanyuGuo, 2024-01-19:
Hello @2892510130, sorry for being so late to get back to you. :pray:

In our project, we used a control plugin called `gazebo_ros_diff_drive`, and you can refer to this [line](https://github.com/ai-winter/ros_motion_planning/blob/7fb18496756c91c04bf524cf791fb12909c985b8/src/sim_env/urdf/turtlebot3_waffle/turtlebot3_waffle.gazebo.xacro#L62) and its [source](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/f9e1a4607842afa5888ef01de31cd64a1e3e297f/gazebo_plugins/src/gazebo_ros_diff_drive.cpp).

If you want to write your own controller in ROS, maybe you can refer to [ros_controllers](https://wiki.ros.org/ros_controllers).

Unfortunately, we don't have any experience developing controllers in ROS. Generally speaking, after getting some higher-level control inputs (such as velocity, acceleration, etc.), we will control motors through a lower computer.

closed 2024-01-19 by ZhanyuGuo