ai-winter/ros_motion_planning #33: Add pedestrians

Suzukitakeko, 2023-10-30:
My configuration file is set like this：
[image]
[image]
But no pedestrians or other dynamic obstacles appear in gazebo and rviz. How do I add them?

ai-winter, 2023-10-31:
Hello, @Suzukitakeko 

I used the following configuration and retested the system.
```yaml
map: "warehouse"
world: "warehouse"
rviz_file: "sim_env.rviz"

robots_config:
  - robot1_type: "turtlebot3_burger"
    robot1_global_planner: "astar"
    robot1_local_planner: "dwa"
    robot1_x_pos: "0.0"
    robot1_y_pos: "0.0"
    robot1_z_pos: "0.0"
    robot1_yaw: "-1.57"

plugins:
  pedestrians: "pedestrian_config.yaml"
```

The results of the run are as follows, showing that pedestrians were correctly detected.

[image]

I assume that your rviz does not have the visualization display for pedestrians added. Please refer to the configuration in `sim_env.rviz` for guidance.

Suzukitakeko, 2023-10-31:
Thanks for your reply, I followed your steps and retested. But there are still no pedestrians here.
[image]
[image]

ai-winter, 2023-11-01:
@Suzukitakeko 

Please provide more details about your system. Additionally, does the terminal output any error messages? Can you obtain data from the `ped_visualization` topic using `rostopic`?

arthuregood, 2023-11-12:
hey @Suzukitakeko , did you install python-is-python3 package?
I was having the same problem, because the main.sh script runs on python - not python3
So you can try to install the package using `sudo apt install python-is-python3` or change the main.sh script :)

Suzukitakeko, 2023-11-12:
@arthuregood ,thank you very much, this is exactly what you said. I successfully solved this problem using your method !

arthuregood, 2023-11-12:
I'm having a related issue tho
My pedestrians get stuck in this loop, no matter when I place them, nor their initial positions and routes
any idea on how to make this work @ai-winter ?
@Suzukitakeko do you have the same problem?

this is my user_config.yaml
```
map: "warehouse"
world: "warehouse"
rviz_file: "sim_env.rviz"

robots_config:
  - robot1_type: "turtlebot3_waffle"
    robot1_global_planner: "a_star"
    robot1_local_planner: "apf_planner"
    robot1_x_pos: "0.0"
    robot1_y_pos: "0.0"
    robot1_z_pos: "0.0"
    robot1_yaw: "0.0"

plugins:
  pedestrians: "pedestrian_config.yaml"
```
this is my pedestrian_config.yaml (the same from README)
```
## pedestrians_config.yaml

# sfm algorithm configure
social_force:
  animation_factor: 5.1
  # only handle pedestrians within `people_distance`
  people_distance: 6.0
  # weights of social force model
  goal_weight: 2.0
  obstacle_weight: 80.0
  social_weight: 15
  group_gaze_weight: 3.0
  group_coh_weight: 2.0
  group_rep_weight: 1.0

# pedestrians setting
pedestrians:
  update_rate: 5
  ped_property:
    - name: human_1
      pose: 5 -2 1 0 0 1.57
      velocity: 0.9
      radius: 0.4
      cycle: true
      time_delay: 5
      ignore:
        model_1: ground_plane
        model_2: turtlebot3_waffle
      trajectory:
        goal_point_1: 5 -2 1 0 0 0
        goal_point_2: 5 2 1 0 0 0
    - name: human_2
      pose: 6 -3 1 0 0 0
      velocity: 1.2
      radius: 0.4
      cycle: true
      time_delay: 3
      ignore:
        model_1: ground_plane
        model_2: turtlebot3_waffle
      trajectory:
        goal_point_1: 6 -3 1 0 0 0
        goal_point_2: 6 4 1 0 0 0
```

https://github.com/ai-winter/ros_motion_planning/assets/53200251/5ec88c85-5563-4769-aeb2-896e289c9efd

Suzukitakeko, 2023-11-12:
@arthuregood Unfortunately I did not encounter the same problem, after solving my original problem the program started to work normally .

arthuregood, 2023-11-12:
could you share your user config, pedestrian config and world config files, please?

arthuregood, 2023-11-12:
@ai-winter I'm going to open a new issue about this, I reverted my repository to the commit f9d52d4 and it works now
probably something that was implemented in the meantime broke the pedestrians simulation

closed 2023-11-13 by ai-winter

ai-winter, 2023-11-13:
Hello, @Suzukitakeko , @arthuregood 

- _Problem 1_: No pedestrians displayed.

As previously mentioned by @arthuregood , please install dependency `python-is-python3`. For more information on dependencies, please refer to the [README.md](https://github.com/ai-winter/ros_motion_planning/blob/master/README.md).

- _Problem 2_：Pedestrians exhibit erroneous movements, and the visualization in Rviz fails.

We apologize for the insufficient testing of recent updates. The bug has been fixed now.

commit a4bb7bef (2023-11-13): fix ed #33 in pedestrian plugins, and closes #33