ai-winter/ros_motion_planning #36: catkin_make error

bigpigdog, 2023-12-10:
...
[ 98%] Linking CXX shared library /home/lhj/motionplanning_ws/devel/lib/libPedestrianSFMPlugin.so
/usr/bin/ld: /usr/local/lib/libprotobuf.a(arena.o): relocation R_X86_64_TPOFF32 against symbol `_ZN6google8protobuf8internal9ArenaImpl13thread_cache_E' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libprotobuf.a(common.o): relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [ros_motion_planning/src/third_party/gazebo_plugins/pedestrian_sfm_plugin/CMakeFiles/PedestrianSFMPlugin.dir/build.make:167: /home/lhj/motionplanning_ws/devel/lib/libPedestrianSFMPlugin.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:5398: ros_motion_planning/src/third_party/gazebo_plugins/pedestrian_sfm_plugin/CMakeFiles/PedestrianSFMPlugin.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

Thank you for your program! The above is my compilation error.
There was no such error a few weeks ago. But today when I moved this project to a new workspace, he compiled with this error. This error occurs in both new and old workspaces, and I haven't fixed the code.

ZhanyuGuo, 2023-12-12:
Hello @bigpigdog, and thanks for your supporting!

We didn't reproduce your problem, did you solve it? If it still doesn't work out and may need you to provide more details, thank you.

bigpigdog, 2023-12-12:
> Hello @bigpigdog, and thanks for your supporting!
> 
> We didn't reproduce your problem, did you solve it? If it still doesn't work out and may need you to provide more details, thank you.

Thanks for your reply!

I think this error has something to do with my system environment, not the project. Because none of my projects compile successfully, including yours, unless I delete the **“gazebo plugin”** from my project. Now I just can't do the simulation.

I deleted all the files in the build and devel folders and recompiled them, and I got this error. I reinstall libprotobuf and add “-fPIC” to CMakeLists based on the error prompt, but it still doesn't work out. Do you have any ideas? Thank you.

sumei666, 2023-12-13:
> > 你好@bigpigdog，并感谢您的支持！
> > 我们没有重现您的问题，您解决了吗？如果仍然无法解决并且可能需要您提供更多详细信息，谢谢。
> 
> 感谢您的回复！
> 
> 我认为这个错误与我的系统环境有关，而不是与项目有关。因为我的项目都没有编译成功，包括你的项目，除非我从我的项目中删除**“gazebo 插件”**。现在我无法进行模拟。
> 
> 我删除了 build 和 devel 文件夹中的所有文件并重新编译它们，然后出现此错误。我重新安装了libprotobuf，并根据错误提示将“-fPIC”添加到CMakeLists中，但仍然不行。你有什么
我也遇到这个问题了 请问怎么解决呢？

bigpigdog, 2023-12-13:
你好@sumei666 ，暂时没有解决方法。我以前是可以用gazebo plugin的，但删了build和devel且重新编译出现这个错误之后就不行了（也许是其他的操作，但我不记得了）。你可以详细说一下你遇到的问题，或许我们可以探讨一下解决方法。

sumei666, 2023-12-13:
> 嗨好@sumei666，暂时没有解决方法。我以前是可以用gazebo插件的，但是删掉了build和devel并且重新编译出现这个错误之后就不行了（也许是其他的操作，但我不记得了）。你可以详细说一下你遇到的问题，或许我们可以探讨一下解决方法。

> 你好@sumei666 ，暂时没有解决方法。我以前是可以用gazebo plugin的，但删了build和devel且重新编译出现这个错误之后就不行了（也许是其他的操作，但我不记得了）。你可以详细说一下你遇到的问题，或许我们可以探讨一下解决方法。

方便给一下WeChat吗？

sumei666, 2023-12-13:
> 你好@sumei666 ，暂时没有解决方法。我以前是可以用gazebo plugin的，但删了build和devel且重新编译出现这个错误之后就不行了（也许是其他的操作，但我不记得了）。你可以详细说一下你遇到的问题，或许我们可以探讨一下解决方法。

the same to you

bigpigdog, 2023-12-13:
@sumei666 My email is in my profile.

bigpigdog, 2023-12-22:
reinstall protobuf 3.6.1(The version depends on your system) solved it. So closed.

1、romove protobuf
```
sudo rm /usr/local/bin/protoc
sudo rm -rf /usr/local/include/google
sudo rm -rf /usr/local/lib/libproto*
sudo rm -rf /usr/lib/protoc
```

2、git clone protobuf 3.6.1

3、git clone googletest-release-1.5.0(The version depends on protobuf). Put googletest-release-1.5.0 into the protobuf/third_party folder, delete the original googletest, and name googletest-release-1.5.0 googletest

4、
```
cd protobuf/
./autogen.sh
./configure --prefix=/usr/local/protobuf CFLAGS="-fPIC"  CXXFLAGS="-fPIC"
make -j12
sudo make install
sudo ldconfig
```

5、Check if the installation is successful
`protoc --version`

closed 2023-12-22 by bigpigdog