ai-winter/ros_motion_planning #41: Another map

muratokusluk, 2024-01-08:
Hello,

I would like to these algorithms my own world. Can you tell me what I have to do? 
I created world and map according to gmapping and rviz file. I use turtlebot3 waffle. Thank you for your supports.

ai-winter, 2024-01-08:
Hello, @muratokusluk 

Please follow the [configuration](https://github.com/ai-winter/ros_motion_planning/blob/master/docs/configuration.md) document to change the `map` option.

muratokusluk, 2024-01-09:
I handle map file. Thank you. But now I got errors when I chose to 2D nav goal.  What am I doing wrong?

[ERROR] [1704830358.782024235, 3206.997000000]: Failed to get a path.
[ERROR] [1704830358.900934445, 3207.100000000]: Failed to get a path.
[ERROR] [1704830359.017752799, 3207.211000000]: Failed to get a path.
[ERROR] [1704830359.111237220, 3207.289000000]: Failed to get a path.
[ERROR] [1704830359.225753013, 3207.395000000]: Failed to get a path.
[ERROR] [1704830359.339621179, 3207.507000000]: Failed to get a path.
[ERROR] [1704830359.442339812, 3207.596000000]: Failed to get a path.
[ERROR] [1704830359.553086547, 3207.701000000]: Failed to get a path.
[ERROR] [1704830359.656893758, 3207.798000000]: Failed to get a path.
[ WARN] [1704830359.747015898, 3207.883000000]: Rotate recovery behavior started.
[ERROR] [1704830366.803094682, 3214.399000000]: Failed to get a path.
[ERROR] [1704830366.907634280, 3214.498000000]: Failed to get a path.
[ERROR] [1704830367.013929708, 3214.597000000]: Failed to get a path.
[ERROR] [1704830368.205782132, 3215.723000000]: Failed to get a path.
[ERROR] [1704830368.891889679, 3216.392000000]: Failed to get a path.
[ERROR] [1704830368.918655234, 3216.417000000]: Failed to get a path.
[ERROR] [1704830369.016950363, 3216.503000000]: Failed to get a path.
[ERROR] [1704830369.115881854, 3216.600000000]: Failed to get a path.
[ERROR] [1704830369.225033723, 3216.706000000]: Failed to get a path.
[ERROR] [1704830369.328452318, 3216.800000000]: Failed to get a path.
[ERROR] [1704830369.429938800, 3216.897000000]: Failed to get a path.
[ERROR] [1704830369.539915518, 3217.002000000]: Failed to get a path.
[ERROR] [1704830369.637656495, 3217.098000000]: Failed to get a path.
[ERROR] [1704830369.752792875, 3217.204000000]: Failed to get a path.
[ERROR] [1704830369.852253753, 3217.299000000]: Failed to get a path.
[ERROR] [1704830369.973568786, 3217.403000000]: Failed to get a path.
[ERROR] [1704830370.086818047, 3217.515000000]: Failed to get a path.
[ERROR] [1704830370.182765852, 3217.614000000]: Failed to get a path.
[ERROR] [1704830370.273904179, 3217.702000000]: Failed to get a path.
[ERROR] [1704830370.378910423, 3217.802000000]: Failed to get a path.
[ERROR] [1704830370.484822318, 3217.907000000]: Failed to get a path.
[ERROR] [1704830370.579542554, 3217.999000000]: Failed to get a path.
[ERROR] [1704830370.686654139, 3218.106000000]: Failed to get a path.
[ERROR] [1704830370.792321131, 3218.204000000]: Failed to get a path.
[ERROR] [1704830370.896137860, 3218.307000000]: Failed to get a path.
[ERROR] [1704830370.993171797, 3218.401000000]: Failed to get a path.
[ERROR] [1704830371.096758836, 3218.501000000]: Failed to get a path.
[ERROR] [1704830371.199381606, 3218.601000000]: Failed to get a path.
[ERROR] [1704830371.293951958, 3218.695000000]: Failed to get a path.
[ERROR] [1704830371.401329405, 3218.799000000]: Failed to get a path.
[ERROR] [1704830371.502513371, 3218.895000000]: Failed to get a path.
[ERROR] [1704830371.610018387, 3218.996000000]: Failed to get a path.
[ERROR] [1704830371.719156768, 3219.101000000]: Failed to get a path.
[ERROR] [1704830371.818638701, 3219.195000000]: Failed to get a path.
[ERROR] [1704830371.927192710, 3219.298000000]: Failed to get a path.
[ERROR] [1704830372.028787335, 3219.391000000]: Failed to get a path.
[ERROR] [1704830372.123331119, 3219.485000000]: Aborting because a valid plan could not be found. Even after executing all recovery behaviors

ai-winter, 2024-01-10:
@muratokusluk  Please provide more details about the map you used and navigation process.

muratokusluk, 2024-01-11:
@ai-winter,

First I created ".world" file and then cloned the turtlebot3 repo on my PC. I opened 3 terminals. 

First terminal I enter this command --> roslaunch flat_terrain_sim flat_terrain_map.launch --> Just opened world in Gazebo
Second terminal I enter: roslaunch turtlebot3_slam turtlebot3_slam.launch
Third Terminal I enter: roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

I moved my robot on the map and created it according to gmappping algorithm. Then opened another terminal and enter this command: rosrun map_server map_saver -f ~/workspace/src/flat_terrain_sim/maps/terrain

After that command generated ".yaml" and ".pgm" files. 
-------------------------------------------------------------------------------------------------------------------
I cloned your repo, created terrain folder at sim_env/maps/terrain, and copied ".yaml" and ".pgm"
I copied ".world" file at sim_env/worlds/

Enter that command: catkin_make
I modified user_config.yaml like 

map: "terrain"
world: "terrain"
rviz_file: "sim_env.rviz"

robots_config:
  - robot1_type: "turtlebot3_waffle"
    robot1_global_planner: "a_star"
    robot1_local_planner: "pid"
    robot1_x_pos: "0.0"
    robot1_y_pos: "0.0"
    robot1_z_pos: "0.0"
    robot1_yaw: "0.0"

Maybe, I should execute roslaunch turtlebot3_navigation turtlebot3_navigation.launch for localization (amcl). I just want to see algorithms on my map. What do I have to do?  Thank you for your support.

muratokusluk, 2024-01-11:
image: /home/murat/workspace/src/flat_terrain_sim/maps/terrain.pgm
resolution: 0.050000
origin: [-10.000000, -10.000000, 0.000000]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.196

This is my map .yaml file

ai-winter, 2024-01-12:
@muratokusluk , so the system can operate normally in the default `warehouse` but fails on your custom map? If that's the case, please provide us with the map for further inspection.

muratokusluk, 2024-01-13:
Hi,

Yes, thats the case. Here you are, thanks

Yang Haodong ***@***.***>, 12 Oca 2024 Cum, 05:23 tarihinde
şunu yazdı:

muratokusluk, 2024-01-16:
Hi @ai-winter,

Could you see my files I sent you? If you see, do you have any idea? Thanks

ai-winter, 2024-01-18:
@muratokusluk 

I have not received any files from you.

muratokusluk, 2024-01-18:
[flat_terrain_sim.zip](https://github.com/ai-winter/ros_motion_planning/files/13980546/flat_terrain_sim.zip)

muratokusluk, 2024-01-18:
Hi @ai-winter,

I sent the files. Thank you

ai-winter, 2024-01-19:
@muratokusluk 
 
I am unable to properly launch the simulation using your `.world` file because I am missing the necessary Gazebo models, such as `tree_7`. Please provide all the files at once to avoid wasting time.

muratokusluk, 2024-01-19:
[models.zip](https://github.com/ai-winter/ros_motion_planning/files/13984459/models.zip)
Sorry, All tree models are in the .zip file. Thanks

ai-winter, 2024-01-19:
@muratokusluk 

s shown in the figure, the system can operate normally. The reason for the inability to plan a path is that the obstacles are too closely spaced. Please modify the map.

[image]

muratokusluk, 2024-01-19:
Thanks @ai-winter