ai-winter/ros_motion_planning #47: RRT not visible on screen

babypisellino, 2024-01-27:
Hi, I was trying to run the code with the RRT algorithm. I edited the file "user_config.yaml" by entering:
robot1_global_planner: "rrt"
robot1_local_planner: "pid"

The only thing that is modified besides this was the gui of gazebo.
Once I run the command on Rviz, using a 2dnav goal, I get nothing. The turtlebot starts moving but I don’t see the RRT tree on the screen. With other global planners instead I see on screen what I should get without problems.
Thanks in advance for the support

ai-winter, 2024-01-27:
@babypisellino 

Please check if the marker data published in the `SamplePlanner/tree` topic has been checked in Rviz.

ZhanyuGuo, 2024-01-27:
@babypisellino, like this.

[image]

closed 2024-04-07 by ai-winter