ai-winter/ros_motion_planning #48: Build Error

ZhanyuGuo, 2024-01-27:
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libignition-math6.so.6.15.0', needed by '/home/gzy/Workspace/ros_motion_planning/devel/.private/gazebo_ped_visualizer_plugin/lib/libPedestrianVisualPlugin.so'.  Stop.

ZhanyuGuo, 2024-01-27:
> make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libignition-math6.so.6.15.0', needed by '/home/gzy/Workspace/ros_motion_planning/devel/.private/gazebo_ped_visualizer_plugin/lib/libPedestrianVisualPlugin.so'. Stop.

[image]
The ignition is upgraded nowadays in ros...

ZhanyuGuo, 2024-01-27:
1. Download source code for ignition-math6_6.15.0.
```bash
git clone https://github.com/gazebosim/gz-math.git -b ignition-math6_6.15.0
```
2. Build ignition.
```bash
cd gz-math/
mkdir build
cd build/
cmake ..
make -j
sudo make install
```
3. Make a symbolic link.
```bash
sudo ln -s /usr/local/lib/libignition-math6.so.6.15.0 /usr/lib/x86_64-linux-gnu/libignition-math6.so.6.15.0
```

ZhanyuGuo, 2024-01-27:
This issue will remain open until `gazebo_ped_visualizer_plugin` version dependency bug on ignition is resolved (I can't find a place to change the version dependencies at the moment).

yangla0701, 2024-05-18:
亲爱的作者大人，只运行RRT算法需要执行什么指令呀

ZhangHan0301, 2024-07-01:
@ZhanyuGuo     at CMakeList.txt       add_compile_options(-std=c++17)

OkitaAqukinn, 2024-08-03:
[image]
HI, seems no method to solve this issue

Cayennexyz, 2024-09-07:
[image]
how to solve this problem

ai-winter, 2024-09-07:
@Cayennexyz 
Please follow README.md and install `conan` properly. Please initiate a new issue if you have any additional questions.

zhangzw2021, 2024-09-20:
> ![erro](https://private-user-images.githubusercontent.com/147065515/365351199-628f8894-3e7d-453c-8617-a3ffdec82dc1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY4MTQ4NDEsIm5iZiI6MTcyNjgxNDU0MSwicGF0aCI6Ii8xNDcwNjU1MTUvMzY1MzUxMTk5LTYyOGY4ODk0LTNlN2QtNDUzYy04NjE3LWEzZmZkZWM4MmRjMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMFQwNjQyMjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNjc5NjI5YWE2YjI3MjY4MTE1ZWZlOGM5ZDliNjY4ZjA1N2Q4MzlkZDU5MjM2NjU5NWI3NGNjNmI2ZGQ1M2M3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.P8B5oREIIdMCByoSjKZusnmUq73AjP8_B8GPR_LCk20) how to solve this problem

do you slove this problem?

WereSongs, 2024-10-30:
> ![erro](https://private-user-images.githubusercontent.com/147065515/365351199-628f8894-3e7d-453c-8617-a3ffdec82dc1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

in the file folder "3rd", run conan_install.sh

lyy-aech, 2024-11-24:
thanks，I have solved this problem.

Mimiaowu, 2024-12-13:
No transform from [base_footprint] to [map].
how to sovle this problem

liusumengshang, 2024-12-24:
[image]
解决方法：

回到src的上一级目录下，运行代码：catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3

bobby-felt, 2024-12-26:
how to solve this issue

bobby-felt, 2024-12-26:
[image]

ZhanyuGuo, 2024-12-26:
Please upload the screenshot again. @bobby-felt

wc123321123, 2025-03-13:
> [image]如何解决这个问题

How did you solve this problem?

akane020915, 2025-03-20:
[image]
how to solve this problem

lzt-888, 2025-03-31:
[image]
[image]   My original intention was to compile the ROS Motion Planning library/ Build.sh encountered difficulties, is there anyone kind-hearted who can help solve them?

lzt-888, 2025-03-31:
> [image]如何解决这个问题

Have you already solved it？

ZhanyuGuo, 2025-04-02:
In addition to OSQP, are there any other 3rd party libraries that need to be managed by `conan`? 

Or should we just use `git submodule` instead? @ai-winter 

Or try to reproduce and solve this issue.

qp-123-qq, 2025-04-04:
[image]这怎么解决啊

qz0122, 2025-04-06:
> [image]这怎么解决啊

也是遇到了这个问题 同学你有解决吗

PhilCartman, 2025-07-26:
The the ubuntu 20.04 default protoc version is 3.4.1, however this project needs protoc == 3.6.1, this problem lead to compile failure
you need update protoc.
 
protoc --version
libprotoc 3.6.1

[image]

PhilCartman, 2025-07-26:
@OkitaAqukinn  It seems to be a problem with the virtual machine