ai-winter/ros_motion_planning #49: SyntaxError: invalid syntax

robot-WT, 2024-01-27:
hello, I encoutered this problem, but it does not seem to affect the navigation. I am not sure if there is an unknown impact. Can you help check it out? thank you.

   File "../src/third_party/dynamic_xml_config/main_generate.py", line 20
     def __init__(self, *plugins) -> None:
                                  ^
SyntaxError: invalid syntax

ai-winter, 2024-01-27:
@robot-WT 

Please select Python 3 to run the program. The failure of this will affect the planning algorithm switching.

robot-WT, 2024-01-27:
yes, you are right. I can't complete the algorithm switch, but after using python3, I still have the above error.

robot-WT, 2024-01-27:
sorry，I was careless. There was one thing I didn’t change. Thank you very much for your answer.
​

closed 2024-01-27 by robot-WT