ai-winter/ros_motion_planning #52: robot not moving through service call

Aak02ash, 2024-02-28:
Hi, I am using RRT* as my global planner and apf as my local planner. When I use 2dNavGoal in rviz, the robot is moving but when i do a service call using 
_rosservice call /move_base/SamplePlanner/make_plan "start: {header: {stamp: now, frame_id: 'map'}, pose: {position: {x: 0.0, y: 0.0, z: 0.0}, orientation: {x: 0.0, y: 0.0, z: 0.0, w: 1.0}}}
goal: {header: {stamp: now, frame_id: 'map'}, pose: {position: {x: -6.4, y: 0.3, z: 0.0}, orientation: {x: 0.0, y: 0.0, z: 0.0, w: 1.0}}}"_

I am getting the plan and tree in rviz but the robot is not moving.
I am not sure whether the goal is passed to move_base. Also I can't find the section of the code where the goal is passed to move_base. Some help is very much appreciated. Thanks in advance.

ai-winter, 2024-02-29:
@Aak02ash 
In fact, you need to pass the goal data to `move_base` through the `move_base_simple/goal` topic.

ai-winter, 2024-02-29:
@Aak02ash 

No source code modification is required. Using the `2D Nav` in Rviz will automatically publish the `move_base_simple/goal` topic and achieve navigation without any additional operations.

Aak02ash, 2024-02-29:
@ai-winter I want to navigate without 2D Nav in Rviz. I wanted to enter a custom goal through terminal. That's why I was doing the service call. But the robot is not moving when I give the goal through service call. Ca you tell me whether there is any other way to give custom goal through terminal?

ai-winter, 2024-02-29:
@Aak02ash  

Like this
```shell
rostopic pub move_base_simple/goal ...
```

closed 2024-03-01 by ai-winter