ai-winter/ros_motion_planning #56: change the algorithm

1littlestudy, 2024-03-28:
Hello, I am a beginner and I would like to ask how to change the algorithm for small cars from A * to rrt algorithm,Thank you in advance for your answer

ai-winter, 2024-03-28:
Hello, @1littlestudy 
`README.md` gives an example in detail, you can also refer to this configure.
```yaml
map: "warehouse"
world: "warehouse"
robots_config:
  - robot1_type: "turtlebot3_burger"
    robot1_global_planner: "astar"
    robot1_local_planner: "dwa"
    robot1_x_pos: "0.0"
    robot1_y_pos: "0.0"
    robot1_z_pos: "0.0"
    robot1_yaw: "-1.57"
  - robot2_type: "turtlebot3_burger"
    robot2_global_planner: "jps"
    robot2_local_planner: "pid"
    robot2_x_pos: "-5.0"
    robot2_y_pos: "-7.5"
    robot2_z_pos: "0.0"
    robot2_yaw: "0.0"
robots_init: "robots_rviz_init.yaml"
rviz_file: "sim_env.rviz"
```

For example, you can change the option `robot1_global_planner` from `astar` to `rrt` to meet your needs

1littlestudy, 2024-03-28:
@ai-winter Thank you very much for your help，I want to ask if it is necessary for me to try changing all the global plans in the image? Can I only modify a few files? such as the start and user without move_base and environment.I would like to ask which files need to be modified to change the planning algorithm?
[image]

ai-winter, 2024-03-28:
@1littlestudy 

You just need to modify the relevant configurations in the `user_config.yaml` file.

1littlestudy, 2024-03-28:
@ai-winter Your code is very beneficial to me. Wishing you success in your future work.

closed 2024-03-28 by ai-winter