ai-winter/ros_motion_planning #58: costmap

zvemif, 2024-04-04:
Hello. I have been puzzled for several days. When using RRT, why can I see the costmap in Rviz simulation, but the path planning does not follow the costmap, instead, it follows the static map layer? Thank you.

ai-winter, 2024-04-07:
@zvemif 

Please attach pictures about your questions.

zvemif, 2024-04-07:
[image]

zvemif, 2024-04-07:
为什么不是按照代价地图进行规划呢？
[image]

ai-winter, 2024-04-07:
The planned path ensures avoidance of high-cost areas; the path depicted in the figure is acceptable. Strict adherence to cost map planning may result in no feasible solutions.

closed 2024-04-07 by ai-winter

zvemif, 2024-04-07:
Thank you! Where can I set it, do I follow the high cost area or set the cost size?

zvemif, 2024-04-07:
> The planned path ensures avoidance of high-cost areas; the path depicted in the figure is acceptable. Strict adherence to cost map planning may result in no feasible solutions.

Thank you! Where can I set up and plan according to the cost map？