ai-winter/ros_motion_planning #62: Display Path

1littlestudy, 2024-04-08:
hi，how to display the path and length of the path that the robot has walked using the RRT algorithm？Thank you in advance for your reply
[image]

githubzhouzz, 2024-04-10:
You can see the blogger's post by displaying the path: https://github.com/ai-winter/ros_motion_planning_visualizer

1littlestudy, 2024-04-11:
Thanks，I downloaded the blogger's code, how to run the result interface。

githubzhouzz, 2024-04-11:
Hello. Previous code when I conducted failed. Now I found a code better.

Link here：
https://blog.csdn.net/qq_45252077/article/details/133824213?ops_request_misc=%257B%2522request%255Fid%2522%253A%2522171284103516800184157738%2522%252C%2522scm%2522%253A%252220140713.130102334..%2522%257D&amp;request_id=171284103516800184157738&amp;biz_id=0&amp;utm_medium=distribute.wap_search_result.none-task-blog-2~all~sobaiduend~default-1-133824213-null-null.wap_first_rank_v2_rank_v29&amp;utm_term=%E5%A4%9A%E6%9C%BA%E5%99%A8%E4%BA%BA%E4%B8%89%E8%A7%92%E5%BD%A2%E7%BC%96%E9%98%9F%E7%9A%84%E5%AE%9E%E7%8E%B0&amp;spm=1018.2118.3001.4187

ps：Although I met the same question as the comments, I think this code hopeful.

1littlestudy, 2024-04-11:
@githubzhouzz Thank you, what I mean is that I am unable to use the ros_motion_planning_visualizer it provides. Can you teach me the specific usage method? I am unable to display the image displayed by him after installing it myself.

omigeft, 2024-04-11:
> @githubzhouzz Thank you, what I mean is that I am unable to use the ros_motion_planning_visualizer it provides. Can you teach me the specific usage method? I am unable to display the image displayed by him after installing it myself.

I feel very sorry! ROS Motion Planning Visualizer is still under development and it hasn't been merged into ROS Motion Planning as a plugin. It can currently only be run by cloning the master branch of its repository to local. If you have any problem in running it, you can raise an issue in repository [ROS Motion Planning Visualizer](https://github.com/ai-winter/ros_motion_planning_visualizer).

githubzhouzz, 2024-04-12:
> @githubzhouzz Thank you, what I mean is that I am unable to use the ros_motion_planning_visualizer it provides. Can you teach me the specific usage method? I am unable to display the image displayed by him after installing it myself.

因为我是邮件看到的，我以为你问的是三机器人编队控制问题。如果是链接（https://github.com/ai-winter/ros_motion_planning_visualizer）的代码，我运行下来是没问题的。你能说一下你具体遇到了什么问题吗？

Because I saw it in an email, I thought you were asking about the control of a three robot formation. If it is a link（ https://github.com/ai-winter/ros_motion_planning_visualizer ）The code, I run it without any problems. Can you tell me specifically what problem you encountered?

1littlestudy, 2024-04-12:
> > 谢谢，我的意思是我无法使用它提供ros_motion_planning_visualizer。你能教我具体的使用方法吗？我自己安装后无法显示他显示的图像。
> 
> 因为我是邮件看到的，我以为你问的是三机器人编队控制问题。如果是链接（[https://github.com/ai-winter/ros_motion_planning_visualizer）的代码，我运行下来是没问题的。你能说一下你具体遇到了什么问题吗？](https://github.com/ai-winter/ros_motion_planning_visualizer%EF%BC%89%E7%9A%84%E4%BB%A3%E7%A0%81%EF%BC%8C%E6%88%91%E8%BF%90%E8%A1%8C%E4%B8%8B%E6%9D%A5%E6%98%AF%E6%B2%A1%E9%97%AE%E9%A2%98%E7%9A%84%E3%80%82%E4%BD%A0%E8%83%BD%E8%AF%B4%E4%B8%80%E4%B8%8B%E4%BD%A0%E5%85%B7%E4%BD%93%E9%81%87%E5%88%B0%E4%BA%86%E4%BB%80%E4%B9%88%E9%97%AE%E9%A2%98%E5%90%97%EF%BC%9F)
> 
> 因为我在一封电子邮件中看到它，我以为你是在问三个机器人编队的控制。如果是链接（https://github.com/ai-winter/ros_motion_planning_visualizer）代码，我运行它没有任何问题。你能具体告诉我你遇到了什么问题吗？

thank you,I have successfully run it, but I am wondering why the implementation is so close to the obstacle.