ai-winter/ros_motion_planning #64: Three-robots PathPlanning Failed

githubzhouzz, 2024-04-10:
Hello. I made the modifications according to what you said and found that although three robots can indeed be displayed, clicking on [2D Nav Goal] does not move the robots. Is there a problem with which namespace? I haven't found it yet.
[image]

githubzhouzz, 2024-04-10:
I want to implement simple navigation and following functions for three robots, similar to the following:https://blog.csdn.net/weixin_46127285/article/details/121015453?ops_request_misc=%257B%2522request%255Fid%2522%253A%2522171273471016777224431075%2522%252C%2522scm%2522%253A%252220140713.130102334.pc%255Fall.%2522%257D&request_id=171273471016777224431075&biz_id=0&utm_medium=distribute.pc_search_result.none-task-blog-2~all~first_rank_ecpm_v1~rank_v31_ecpm-1-121015453-null-null.142^v100^pc_search_result_base1&utm_term=%E9%A2%86%E8%88%AA%E8%B7%9F%E9%9A%8F%E6%B3%95%E7%BC%96%E9%98%9F%E6%8E%A7%E5%88%B6ros&spm=1018.2226.3001.4187
What files should I modify?

githubzhouzz, 2024-04-10:
And what files should I modify when I need to set up obstacle avoidance settings for the following robots? (Using PSO for piloting drones,Nash game for the following robots)

githubzhouzz, 2024-04-10:
I modified the 22nd line of the file according to my understanding:
 ros_motion_planning/src/sim_env/launch/include/navigation/movebase. launch.xml, 
[image]
and changed the robot to robot1 (set robot1 as the navigation robot and perform PSO path planning on it), but an error occurred, as shown in the following figure.
[image]

githubzhouzz, 2024-04-10:
I tried using the "rostopic pub" and it popped up with "publishing and latching message. Press ctrl C to terminate" and the robot did not move.

ros@ros-virtual-machine:~$ rostopic pub /robot1/move_base/EvolutionaryPlanner/pso_marker visualization_msgs/Marker "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
ns: ''
id: 0
type: 0
action: 0
pose:
  position: {x: 2.0, y: 5.0, z: 0.0}
  orientation: {x: 0.0, y: 0.0, z: 0.0, w: 0.0}
scale: {x: 0.0, y: 0.0, z: 0.0}
color: {r: 255.0, g: 0.0, b: 0.0, a: 0.0}
lifetime: {secs: 0, nsecs: 0}
frame_locked: false
points:
- {x: 0.0, y: 0.0, z: 0.0}
colors:
- {r: 0.0, g: 255.0, b: 0.0, a: 0.0}
text: ''
mesh_resource: ''
mesh_use_embedded_materials: false" 
publishing and latching message. Press ctrl-C to terminate

ai-winter, 2024-04-10:
@githubzhouzz 

To enable the correct namespace for multiple robots, it is necessary to set the `rviz_file` in the `user_config.yaml` to `""`. As for the multi-robot navigation task, you are required to implement it based on the existing code by yourself.

githubzhouzz, 2024-04-10:
Okay, thank you! Your code has been very helpful to me, thank you very much!

githubzhouzz, 2024-04-17:
> @githubzhouzz
> 
> To enable the correct namespace for multiple robots, it is necessary to set the `rviz_file` in the `user_config.yaml` to `""`. As for the multi-robot navigation task, you are required to implement it based on the existing code by yourself.

I have modified the file and it can indeed display three robots, but there is an error in the movebase file. Specifically, when using [2D Nav Goal], some of the three robots do not move, while others move randomly.
[image]
[image]

githubzhouzz, 2024-04-18:
I changed the local path planning from PID to DWA, resulting in the following error:

[ERROR] [1713444044.517826715, 18.426000000]: Extrapolation Error: Lookup would require extrapolation 0.002000000s into the future.  Requested time 18.402000000 but the latest data is at time 18.400000000, when looking up transform from frame [robot1/odom] to frame [map]

[ERROR] [1713444044.517873187, 18.426000000]: Global Frame: robot1/odom Plan Frame size 68: map

[ WARN] [1713444044.517903951, 18.426000000]: Could not transform the global plan to the frame of the controller
[ERROR] [1713444044.517924455, 18.426000000]: Could not get local plan

githubzhouzz, 2024-04-18:
When I used pid,apf,rpp,lqr,static, all generated the same errors. Only dwa generated different errors and the least.(When viewed through the ROSWTF statement, the display shows the least number of errors.)