ai-winter/ros_motion_planning #65: Failed to run MPC local planner

bigpigdog, 2024-04-11:
I've installed all the dependencies, but this error occurs:

`[FATAL] [1712800952.056223995, 1711543538.130066234]: Failed to create the mpc_planner/MPCPlanner planner, are you sure it is properly registered and that the containing library is built? Exception: Failed to load library /home/lhj/catkin_ws/devel/lib//libmpc_planner.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libOsqpEigen.so.0.8.1: cannot open shared object file: No such file or directory)`

The `libmpc_planner.so` and `libOsqpEigen.so.0.8.1` does exist. Why compiler show that `libmpc_planner.so` exist in `***/catkin_ws/devel/lib//libmpc_planner.so` and not in `***/catkin_ws/devel/lib/libmpc_planner.so`. In the latter, there are no two "/" that are the right path.

bigpigdog, 2024-04-11:
Reinstalling osqp-eigen and osqp solved the issue. But I reinstalled it once before. It's very weird.

But another problem occurs. 

`ERROR in LDL_factor: Error in KKT matrix LDL factorization when computing the nonzero elements. The problem seems to be non-convex
ERROR in osqp_setup: KKT matrix factorization.
The problem seems to be non-convex.
[OsqpEigen::Solver::initSolver] Unable to setup the workspace.
[OsqpEigen::Solver::solveProblem] The solve has not been initialized yet. Please call initSolver() method.`

Next I'm going to read the code and try to fix the problem...

ai-winter, 2024-04-18:
@bigpigdog 

We did not encounter this issue during testing. If possible, could you please try to reproduce the error and provide the relevant log files? We will attempt to identify and fix the issue.

bigpigdog, 2024-04-23:
@ai-winter 

Thanks for your reply! I don't know which is the relevant log, so here I provide some logs.

**roslaunch-WP-111125.log**
```
[roslaunch][INFO] 2024-04-23 15:09:28,454: ... preparing to launch node of type [move_base/move_base]
[roslaunch][INFO] 2024-04-23 15:09:28,454: create_node_process: package[move_base] type[move_base] machine[Machine(name[] env_loader[None] address[localhost] ssh_port[22] user[None] assignable[True] timeout[10.0])] master_uri[http://localhost:11311]
[roslaunch][INFO] 2024-04-23 15:09:28,454: process[move_base-6]: env[{'SHELL': '/bin/bash', 'ROS_VERSION': '1', 'SESSION_MANAGER': 'local/WP:@/tmp/.ICE-unix/2155,unix/WP:/tmp/.ICE-unix/2155', 'QT_ACCESSIBILITY': '1', 'COLORTERM': 'truecolor', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/etc/xdg', 'XDG_MENU_PREFIX': 'gnome-', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'GTK_IM_MODULE': 'fcitx', 'CONDA_EXE': '/home/lhj/anaconda3/bin/conda', '_CE_M': '', 'PKG_CONFIG_PATH': '/home/lhj/catkin_mpc_ws/devel/lib/pkgconfig:/home/lhj/catkin_pp_ws/devel/lib/pkgconfig:/home/lhj/catkin_google_ws/devel_isolated/cartographer_rviz/lib/pkgconfig:/home/lhj/catkin_google_ws/install_isolated/lib/pkgconfig:/home/lhj/hdl_localization_ws/devel/lib/pkgconfig:/home/lhj/lio_relocalization_ws/devel/lib/pkgconfig:/home/lhj/catkin_ws/devel/lib/pkgconfig:/opt/ros/noetic/lib/pkgconfig:/opt/ros/noetic/lib/x86_64-linux-gnu/pkgconfig', 'ROS_PYTHON_VERSION': '3', 'TERMINATOR_DBUS_PATH': '/net/tenshu/Terminator2', 'QT4_IM_MODULE': 'fcitx', 'LC_ADDRESS': 'zh_CN.UTF-8', 'GNOME_SHELL_SESSION_MODE': 'ubuntu', 'LC_NAME': 'zh_CN.UTF-8', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'TERMINATOR_UUID': 'urn:uuid:bfa6b013-6cb1-40ec-a9cd-06a7ba0b44b7', 'XMODIFIERS': '@im=fcitx', 'DESKTOP_SESSION': 'ubuntu', 'LC_MONETARY': 'zh_CN.UTF-8', 'SSH_AGENT_PID': '2109', 'ROS_PACKAGE_PATH': '/home/lhj/catkin_mpc_ws/src:/home/lhj/catkin_pp_ws/src:/home/lhj/catkin_google_ws/src/cartographer_ros/cartographer_rviz:/home/lhj/catkin_google_ws/install_isolated/share:/home/lhj/hdl_localization_ws/src:/home/lhj/lio_relocalization_ws/src:/home/lhj/catkin_ws/src:/opt/ros/noetic/share', 'GTK_MODULES': 'gail:atk-bridge', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/lhj/catkin_mpc_ws/devel/share/common-lisp:/home/lhj/catkin_pp_ws/devel/share/common-lisp:/home/lhj/catkin_google_ws/devel_isolated/cartographer_rviz/share/common-lisp:/home/lhj/hdl_localization_ws/devel/share/common-lisp:/home/lhj/lio_relocalization_ws/devel/share/common-lisp:/home/lhj/catkin_ws/devel/share/common-lisp', 'PWD': '/home/lhj', 'LOGNAME': 'lhj', 'XDG_SESSION_DESKTOP': 'ubuntu', 'XDG_SESSION_TYPE': 'x11', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'XAUTHORITY': '/run/user/1000/gdm/Xauthority', 'WINDOWPATH': '2', 'HOME': '/home/lhj', 'USERNAME': 'lhj', 'IM_CONFIG_PHASE': '1', 'LC_PAPER': 'zh_CN.UTF-8', 'LANG': 'en_US.UTF-8', 'ROS_ETC_DIR': '/opt/ros/noetic/etc/ros', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'XDG_CURRENT_DESKTOP': 'ubuntu:GNOME', 'VTE_VERSION': '6003', 'CMAKE_PREFIX_PATH': '/home/lhj/catkin_mpc_ws/devel:/home/lhj/catkin_pp_ws/devel:/home/lhj/catkin_google_ws/devel_isolated/cartographer_rviz:/home/lhj/catkin_google_ws/install_isolated:/home/lhj/hdl_localization_ws/devel:/home/lhj/lio_relocalization_ws/devel:/home/lhj/catkin_ws/devel:/opt/ros/noetic', 'INVOCATION_ID': 'ae9020b19018442fafca325eff832690', 'TERMINATOR_DBUS_NAME': 'net.tenshu.Terminator23558193cd9818af7fe4d2c2f5bd9d00f', 'MANAGERPID': '1891', 'CLUTTER_IM_MODULE': 'fcitx', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'XDG_SESSION_CLASS': 'user', 'PYTHONPATH': '/home/lhj/catkin_mpc_ws/devel/lib/python3/dist-packages:/home/lhj/catkin_pp_ws/devel/lib/python3/dist-packages:/home/lhj/catkin_google_ws/install_isolated/lib/python3/dist-packages:/home/lhj/hdl_localization_ws/devel/lib/python3/dist-packages:/home/lhj/lio_relocalization_ws/devel/lib/python3/dist-packages:/home/lhj/catkin_ws/devel/lib/python3/dist-packages:/opt/ros/noetic/lib/python3/dist-packages', 'TERM': 'xterm-256color', 'LC_IDENT [... cut ...]
[rosmaster.master][INFO] 2024-04-23 15:09:30,077: publisherUpdate[/move_base_simple/goal] -> http://WP:33315/ ['http://WP:38173/', 'http://WP:43363/']: sec=0.00, result=[1, '', 0]
[rosmaster.master][INFO] 2024-04-23 15:09:30,080: +SERVICE [/rviz/reload_shaders] /rviz http://WP:43363/
[rosmaster.master][INFO] 2024-04-23 15:09:30,080: +SERVICE [/rviz/load_config] /rviz http://WP:43363/
[rosmaster.master][INFO] 2024-04-23 15:09:30,081: +SERVICE [/rviz/load_config_discarding_changes] /rviz http://WP:43363/
[rosmaster.master][INFO] 2024-04-23 15:09:30,081: +SERVICE [/rviz/save_config] /rviz http://WP:43363/
```

bigpigdog, 2024-05-07:
@ai-winter 

用中文交流吧。用 mpc_planner 跑你们的 gazebo demo（main.sh）是没有任何错误的。（上一条发的一些log信息没有意义，我删除了）

> Reinstalling osqp-eigen and osqp solved the issue. But I reinstalled it once before. It's very weird.
>
> But another problem occurs.
>
> `ERROR in LDL_factor: Error in KKT matrix LDL factorization when [...]

我的想法是把你们的 mpc_planner 用于没有全局地图的路径规划中，我修改了 move_base 中 global_planner、local_costmap 以及 global_costmap 的运行方式（不使用全局规划器与全局地图，通过某种算法给定一条无碰撞路径，然后使用mpc跟随），然后遇到了如上的报错。**后来发现是 kappa 有时候太小会出现 nan，然后导致 u_r 和 u 也出现 nan，因此加了一个判断，kappa 为 nan 的时候让它等于0。**除了这个原因，其他时候也会出现，但频率已经非常非常小了，遂没有管他。

**我想知道如何显示 mpc k时刻的预测轨迹？**下面是我添加的代码，我不太清楚是否正确。`solution` 的结果是k时刻对未来k+n步的输出预测，加上 du_p 和 u_r 才是正确的 v 和 w 吗？`X`计算出了未来k+n步的相对坐标，加上 s_d 才是 odom 系下的真实坐标？

绿色线条就是我生成的 trajectory_path。但是感觉很奇怪，起始点不在 base_footprint 的原点上，而且线段距离很长，图中有3m多（dt = 0.05, p = 12, max_v = 0.5），这个参数的预测距离最长应该是 0.05 * 12 * 0.5 = 0.3m 吧？
[image]

```
  for (int i = 0; i < 2 * m_; ++i) 
  {
    if (i % 2 == 0) {
      solution[i] = solution[i] + du_p[0] + u_r[0];
    } else {
      solution[i] = regularizeAngle(solution[i] + du_p[1] + u_r[1]);
    }
  }
  Eigen::VectorXd X = S_x * x + S_u * solution; // (3p x 1)

  nav_msgs::Path trajectory_path;
  trajectory_path.header.stamp = ros::Time::now();
  trajectory_path.header.frame_id = "odom";
  geometry_msgs::PoseStamped pose;
  pose.header.frame_id = trajectory_path.header.frame_id;
  pose.header.stamp = ros::Time::now();
  pose.pose.orientation.w = 1.0;
  pose.pose.position.z = 1.0;
  for (int i = 0; i < p_; i++)
  {
    pose.pose.position.x = s_d[0] + X[i * 3];
    pose.pose.position.y = s_d[1] + X[i * 3 + 1];
    trajectory_path.poses.push_back(pose);
  }
  trajectory_pub_.publish(trajectory_path);
```

ai-winter, 2024-05-07:
@bigpigdog 

感谢你的分析，曲率的问题我们将在未来的工作中改进。关于预测轨迹，可以基于运动学模型向前推理若干时间步并可视化

ai-winter, 2025-01-03:
We configured the dependency for the third-party `OSQP` dynamic library through `Conan` and removed `OSQPEigen`. The latest version does not have this issue, so this issue is now closed.

closed 2025-01-03 by ai-winter