ai-winter/ros_motion_planning #66: Continuously planning the path

1littlestudy, 2024-04-12:
hello，When I use the  RRT-connect algorithm, I am unable to reach the target point. He constantly obtains new paths, and I want to ask how to stop constantly planning. After the car completes a path planning, it travels along the road planned by the RRT algorithm.
[image]
and Can I turn off local programming and only use the RRT algorithm for global programming? If possible, what should I do?

ai-winter, 2024-04-12:
@1littlestudy 

The re-planning in global path planning ensures the robot's adaptability to dynamic obstacles, while local planning ensures the robot's ability to track the planned path. Both are indispensable for effective navigation.

1littlestudy, 2024-04-12:
As you can see, my rrt connect is hovering in place and I do not add dynamic obstacles such as pedestrians. I would like to inquire about how to add dynamic obstacles and how to solve the problem of not being able to reach the target point
[image]

ai-winter, 2024-04-12:
Could you upload a video regarding this error?

ai-winter, 2024-04-12:
Which LOCAL PLANNER did you select?

1littlestudy, 2024-04-12:
I used the PID algorithm

> 您选择了哪个本地规划师？

ai-winter, 2024-04-12:
@1littlestudy 

Could you please test other local planners, such as LQR, RPP, MPC, DWA, and APF, and report on their compatibility with the RRT algorithm?

1littlestudy, 2024-04-12:
> @1littlestudy
> 
> 您能否测试其他本地规划器，例如 LQR、RPP、MPC、DWA 和 APF，并报告它们与 RRT 算法的兼容性？

Okay, I will inform you of the results later. Please give me some time, thank you.

1littlestudy, 2024-04-14:
I'm sorry for only replying to you now. Thank you. I changed the local planning algorithm and successfully reached the target point.
I think the reason for the failure of the rrt connect algorithm on PID is that the refresh rate of the global planning is too fast, causing it to hover in place.
This is because I accidentally modified the initial position of the car, resulting in only one global planning and the rrt connect algorithm successfully reaching the target point.
[image]

Q：I am conducting an RRT comparison experiment using a static map, so I would like to turn off the re planning of global path planning. Can you tell me how to modify it?

1littlestudy, 2024-04-14:
> 很抱歉现在才回复你。谢谢。我更改了本地计划算法并成功到达了目标点。我认为 RRT 连接算法在 PID 上失败的原因是全局规划的刷新率太快，导致它悬停在原地。这是因为我不小心修改了汽车的初始位置，导致只有一个全局规划，rrt 连接算法成功到达目标点。 ![image](https://private-user-images.githubusercontent.com/165125780/322261675-690c2134-05d5-47c9-b3d4-c1d319eb65d2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMwODk1MDksIm5iZiI6MTcxMzA4OTIwOSwicGF0aCI6Ii8xNjUxMjU3ODAvMzIyMjYxNjc1LTY5MGMyMTM0LTA1ZDUtNDdjOS1iM2Q0LWMxZDMxOWViNjVkMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxNFQxMDA2NDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hNWQ5ZmJiOGQzN2MxNjZlMzcyY2RmZmMxZTQ1ZjlmM2M3NWQ5YmNjYzg5NGY2YzdjNjNkZDgxOTZhZDRjMTkzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5Tzv0MdlV_5zHhuP8tmHYbk0G8hUs1N-Gvb51h6No-0)
> 
> Q：我正在使用静态地图进行RRT比较实验，因此我想关闭全局路径规划的重新规划。你能告诉我如何修改它吗？

I tried to modify the frequency in the planning and successfully modified the frequency of the global planning. I am still trying to understand and learn the code. Thank you for your help. I will consult with you if there are any further questions. Thank you.

[image]