ai-winter/ros_motion_planning #68: implementations with move_base

idabble31, 2024-04-16:
Hello, I'm currently exploring the possibility of incorporating the global planner algorithms from this repository into my mobile robot's navigation system, which currently utilizes the ROS Move Base package. Is there a method to integrate these planner algorithms as a plugin within Move Base, or is there another way to achieve this integration?

xiaochoumengyan, 2024-04-17:
> 您好，我目前正在探索将该存储库中的全局规划器算法合并到我的移动机器人导航系统中的可能性，该系统目前使用 ROS Move Base 包。有没有办法将这些规划器算法作为插件集成到 Move Base 中，或者有没有其他方法可以实现这种集成？

请问你现在集成成功啦吗？

omigeft, 2024-04-17:
@idabble31 @xiaochoumengyan For the deployment of algorithms on real robots, you can refer to [https://github.com/ai-winter/ros_motion_planning/blob/master/docs/realworld.md](https://github.com/ai-winter/ros_motion_planning/blob/master/docs/realworld.md). But it is notable that this tutorial only guides simple deployment. More customized deployments require you to delve deeper.

ai-winter, 2024-04-18:
@idabble31 @xiaochoumengyan 

This project is implemented based on the ROS `move_base` framework, making it convenient to port with the help of plugin mechanisms. You can refer to the relevant ROS documents for further guidance.

closed 2024-04-18 by ai-winter