ai-winter/ros_motion_planning #72: Catkin_make error at 93%.

hasel17, 2024-04-26:
I already install dekstop-full ROS Noetic and the needed dependences from readme.md. but catkin_make error at compaling. can anyone help me solve the problem?

here's the log

'
[ 93%] Building CXX object third_party/rviz_plugins/spencer_tracking_rviz_plugin/CMakeFiles/spencer_tracking_rviz_plugin.dir/src/moc_social_activities_display.cpp.o
In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:40,
                 from /home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/person_display_common.h:40,
                 from /home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/social_relations_display.h:36,
                 from /home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/social_relations_display.cpp:36:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:52,
                 from /home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/tracked_persons_cache.h:40,
                 from /home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/tracked_persons_cache.cpp:31:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:40,
                 from /home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/../../../../../src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/person_display_common.h:40,
                 from /home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/../../../../../src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/social_activities_display.h:39,
                 from /home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/moc_social_activities_display.cpp:9:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/../../../../../src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:40,
                 from /home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/../../../../../src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/person_display_common.h:40,
                 from /home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/../../../../../src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/social_relations_display.h:36,
                 from /home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/moc_social_relations_display.cpp:9:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/../../../../../src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:40,
                 from /home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/person_display_common.h:40,
                 from /home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/tracked_groups_display.h:40,
                 from /home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/tracked_groups_display.cpp:35:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:40,
                 from /home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/../../../../../src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/person_display_common.h:40,
                 from /home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/../../../../../src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/tracked_groups_display.h:40,
                 from /home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/moc_tracked_groups_display.cpp:9:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/home/nazrin/ros_motion_planning/build/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/../../../../../src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:40,
                 from /home/nazrin/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/person_display_common.h:40,
[... cut ...] rror 1
make[2]: *** [third_party/rviz_plugins/spencer_tracking_rviz_plugin/CMakeFiles/spencer_tracking_rviz_plugin.dir/build.make:147: third_party/rviz_plugins/spencer_tracking_rviz_plugin/CMakeFiles/spencer_tracking_rviz_plugin.dir/src/social_activities_display.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7528: third_party/rviz_plugins/spencer_tracking_rviz_plugin/CMakeFiles/spencer_tracking_rviz_plugin.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed
'

ai-winter, 2024-05-14:
@Littledragon-wxl 

It seems to be a compatibility issue between QT and Rviz. Please provide the versions of both.

Littledragon-wxl, 2024-05-14:
@ai-winter 
The version of QT is 5.12.8, and the version of Rviz is 1.14.23

ai-winter, 2024-05-16:
@Littledragon-wxl 

Could you try reinstall Rviz and compile again?

ParasRanaa, 2024-05-18:
Is this issue resolved?

MoyuXiansheng, 2024-05-22:
Is this issue resolved?

jusemat, 2024-05-22:
I has a similar problem after update recently the rviz version via apt update command.
I just solve the problem by change the QObject heritance to public in the additional_topic_subscriber.h file
In your case probably must to change the line 63 in _your_workspace/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h_ from "class _AdditionalTopicSubscriber: QObject" to "class _AdditionalTopicSubscriber: **public** QObject"

mohd109, 2024-05-22:
@jusemat Thanks a lot, that solved my issue.

closed 2024-05-24 by ai-winter

XinJingHao, 2024-11-20:
> I has a similar problem with another pkg that use the spencer_tracking_rviz_plugin after update recently the rviz version via apt update command (from version 1.14.20 to 1.14.23). I just solve the problem by change the QObject heritance to public in the additional_topic_subscriber.h file In your case probably must to change the line 63 in _your_workspace/ros_motion_planning/src/third_party/rviz_plugins/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h_ from "class _AdditionalTopicSubscriber: QObject" to "class _AdditionalTopicSubscriber: **public** QObject"

Solved, thanks a lot!

commit 6c130f0e (2024-05-24): fix bugs and close #72