ai-winter/ros_motion_planning #79: There was an error during compilation regarding Gazebo

chan-yuu, 2024-05-28:
Failed     << gazebo_sfm_plugin:make                      [ Exited with code 2 ]
Failed    <<< gazebo_sfm_plugin                           [ 13.3 seconds ]     
Abandoned <<< gazebo_ped_visualizer_plugin                [ Unrelated job failed ]

[image]

how to solve it?

ai-winter, 2024-05-28:
@chan-yuu 

This project should be compiled on Ubuntu 20.04.

chan-yuu, 2024-05-28:
Thank you!
There should be an issue with the version of protobuf.
What is the version of protobuf?

chan-yuu, 2024-05-28:
protoc'version = 3.6.1
Solved it

closed 2024-05-28 by ai-winter

fjc7, 2024-07-10:
Hello,which version of gazebo should be used in this program?
In my computer,the version of gazebo is follow:

(base) jc@jc-e52690v3:~/git_project/ros_motion_planning/scripts$ gazebo --version
Gazebo multi-robot simulator, version 11.11.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

But when I run the build.sh in the ./script, I get a samlier error as follow:

289 |   essid_.SetAllocatedNoArena(&::google::protobuf::internal::GetEmptyStringAlreadyInited(), essid);
      |          ^~~~~~~~~~~~~~~~~~~
      |          SetAllocated
In file included from /usr/include/gazebo-11/gazebo/msgs/MessageTypes.hh:130,
                 from /usr/include/gazebo-11/gazebo/msgs/msgs.hh:35,
                 from /usr/include/gazebo-11/gazebo/physics/Entity.hh:28,
                 from /usr/include/gazebo-11/gazebo/physics/Model.hh:30,
                 from /usr/include/gazebo-11/gazebo/physics/Actor.hh:27,
                 from /usr/include/gazebo-11/gazebo/physics/physics.hh:2,
                 from /home/jc/git_project/ros_motion_planning/src/plugins/gazebo_plugins/pedestrian_sfm_plugin/include/pedestrian_sfm_plugin.h:31,
                 from /home/jc/git_project/ros_motion_planning/src/plugins/gazebo_plugins/pedestrian_sfm_plugin/src/pedestrian_sfm_plugin.cpp:21:
/usr/include/gazebo-11/gazebo/msgs/wireless_nodes.pb.h: At global scope:
/usr/include/gazebo-11/gazebo/msgs/wireless_nodes.pb.h:51:46: error: ‘FieldMetadata’ in namespace ‘google::protobuf::internal’ does not name a type; did you mean ‘AnyMetadata’?
   51 |   static const ::google::protobuf::internal::FieldMetadata field_metadata[];
      |                                              ^~~~~~~~~~~~~
      |                                              AnyMetadata
/usr/include/gazebo-11/gazebo/msgs/wireless_nodes.pb.h:52:46: error: ‘SerializationTable’ in namespace ‘google::protobuf::internal’ does not name a type
   52 |   static const ::google::protobuf::internal::SerializationTable serialization_table[];
      |                                              ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/gazebo-11/gazebo/msgs/MessageTypes.hh:130,
                 from /usr/include/gazebo-11/gazebo/msgs/msgs.hh:35,
                 from /usr/include/gazebo-11/gazebo/physics/Entity.hh:28,
                 from /usr/include/gazebo-11/gazebo/physics/Model.hh:30,
                 from /usr/include/gazebo-11/gazebo/physics/Actor.hh:27,
                 from /usr/include/gazebo-11/gazebo/physics/physics.hh:2,
                 from /home/jc/git_project/ros_motion_planning/src/plugins/gazebo_plugins/pedestrian_sfm_plugin/include/pedestrian_sfm_plugin.h:31,
                 from /home/jc/git_project/ros_motion_planning/src/plugins/gazebo_plugins/pedestrian_sfm_plugin/src/pedestrian_sfm_plugin.cpp:21:
/usr/include/gazebo-11/gazebo/msgs/wireless_nodes.pb.h:183:33: error: ‘InternalMetadataWithArena’ in namespace ‘google::protobuf::internal’ does not name a type; did you mean ‘InternalMetadata’?
  183 |   ::google::protobuf::internal::InternalMetadataWithArena _internal_metadata_;
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 InternalMetadata
/usr/include/gazebo-11/gazebo/msgs/wireless_nodes.pb.h:125:25: error: ‘gazebo::msgs::WirelessNodes* gazebo::msgs::WirelessNodes::New() const’ marked ‘final’, but is not virtual
  125 |   inline WirelessNodes* New() const final {
      |                         ^~~
/usr/include/gazebo-11/gazebo/msgs/wireless_nodes.pb.h:140:8: error: ‘bool gazebo::msgs::WirelessNodes::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)’ marked ‘final’, but is not virtual
  140 |   bool MergePartialFromCodedStream(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/msgs/wireless_nodes.pb.h:142:8: error: ‘void gazebo::msgs::WirelessNodes::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const’ marked ‘final’, but is not virtual
  142 |   void SerializeWithCachedSizes(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/msgs/wireless_nodes.pb.h:144:30: error: ‘google::protobuf::uint8* gazebo::msgs::WirelessNodes::InternalSerializeWithCachedSizesToArray(bool, google::protobuf::uint8*) const’ marked ‘final’, but is not virtual
  144 |   ::google::protobuf::uint8* InternalSerializeWithCachedSizesToArray(
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/msgs/wireless_nodes.pb.h: In member function ‘const google::protobuf::UnknownFieldSet& gazebo::msgs::WirelessNodes::unknown_fields() const’:
/usr/include/gazebo-11/gazebo/msgs/wireless_nodes.pb.h:101:47: error: no matching function for call to ‘google::protobuf::internal::InternalMetadata::unknown_fields() const’
  101 |     return _internal_metadata_.unknown_fields();
      |                                               ^
In file included from /home/jc/anaconda3/include/google/protobuf/message_lite.h:54,
                 from /home/jc/anaconda3/include/google/protobuf/generated_enum_util.h:38,
                 from /home/jc/anaconda3/include/google/protobuf/map.h:60,
                 from /usr/include/google/protobuf/generated_message_table_driven.h:34,
                 from /usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:25,
                 from /usr/include/gazebo-11/gazebo/physics/PhysicsTypes.hh:26,
[... cut ...]
  105 |     return _internal_metadata_.mutable_unknown_fields();
      |                                                       ^
make[2]: *** [plugins/gazebo_plugins/pedestrian_sfm_plugin/CMakeFiles/PedestrianSFMPlugin.dir/build.make:63: plugins/gazebo_plugins/pedestrian_sfm_plugin/CMakeFiles/PedestrianSFMPlugin.dir/src/pedestrian_sfm_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:5526: plugins/gazebo_plugins/pedestrian_sfm_plugin/CMakeFiles/PedestrianSFMPlugin.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j24 -l24" failed

fjc7, 2024-07-10:
Ubuntu 20.04.6+ROS noetic

ai-winter, 2024-07-10:
@fjc7 

It seemsto be a `protobuf` issue rather than a gazebo issue, please check the version of `protobuf`.

fjc7, 2024-07-10:
谢谢，重新安装了下ros，删除了build文件夹，取消conda虚拟环境，编译通过了

DoongLi, 2024-09-11:
@fjc7 感谢你让我看到这个issue，实际上是conda环境的问题，解决方法如下：
'
(base) doongli@doongli-ROG-Strix-G733ZM-G733ZM:~$ conda deactivate
doongli@doongli-ROG-Strix-G733ZM-G733ZM:~$ catkin_make
'