ai-winter/ros_motion_planning #89: [Compile ERROR]

CHB-learner, 2024-07-13:
# 我成功的完成了之前的每一步，版本也没问题。但在编译时报错。请问这个是什么冲突了嘛？

hb@hb:~/Projects_ICT/ros_motion_planning$ cd scripts/
hb@hb:~/Projects_ICT/ros_motion_planning/scripts$ ./build.sh
Configuration:
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.libcxx=libstdc++
compiler.version=9
os=Linux
os_build=Linux
[options]
[build_requires]
[env]

conanfile.py: Installing package
Requirements
    osqp/0.6.3 from 'conancenter' - Cache
Packages
    osqp/0.6.3:6af9cc7cb931c5ad942174fd7838eb655717c709 - Cache

Installing (downloading, building) binaries...
osqp/0.6.3: Already installed!
conanfile.py: WARN: Using the new toolchains and generators without specifying a build profile (e.g: -pr:b=default) is discouraged and might cause failures and unexpected behavior
conanfile.py: Generator 'CMakeToolchain' calling 'generate()'
conanfile.py: Preset 'release' added to CMakePresets.json. Invoke it manually using 'cmake --preset release'
conanfile.py: If your CMake version is not compatible with CMakePresets (<3.19) call cmake like: 'cmake <path> -G "Unix Makefiles" -DCMAKE_TOOLCHAIN_FILE=/home/hb/Projects_ICT/ros_motion_planning/3rd/build/build/Release/generators/conan_toolchain.cmake -DCMAKE_POLICY_DEFAULT_CMP0091=NEW -DCMAKE_BUILD_TYPE=Release'
conanfile.py: Generator cmake_find_package created Findosqp.cmake
conanfile.py: Generator cmake created conanbuildinfo.cmake
conanfile.py: WARN: Using the new toolchains and generators without specifying a build profile (e.g: -pr:b=default) is discouraged and might cause failures and unexpected behavior
conanfile.py: Generator 'CMakeDeps' calling 'generate()'
conanfile.py: Generator txt created conanbuildinfo.txt
conanfile.py: Aggregating env generators
conanfile.py: Generated conaninfo.txt
conanfile.py: Generated graphinfo
Base path: /home/hb/Projects_ICT/ros_motion_planning
Source space: /home/hb/Projects_ICT/ros_motion_planning/src
Build space: /home/hb/Projects_ICT/ros_motion_planning/build
Devel space: /home/hb/Projects_ICT/ros_motion_planning/devel
Install space: /home/hb/Projects_ICT/ros_motion_planning/install
Creating symlink "/home/hb/Projects_ICT/ros_motion_planning/src/CMakeLists.txt" pointing to "/opt/ros/noetic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/hb/Projects_ICT/ros_motion_planning/src -DCATKIN_DEVEL_PREFIX=/home/hb/Projects_ICT/ros_motion_planning/devel -DCMAKE_INSTALL_PREFIX=/home/hb/Projects_ICT/ros_motion_planning/install -G Unix Makefiles" in "/home/hb/Projects_ICT/ros_motion_planning/build"
####
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/hb/Projects_ICT/ros_motion_planning/devel
-- Using CMAKE_PREFIX_PATH: /home/hb/carto_ws/cartographer_detailed_comments_ws-master/install_isolated;/opt/ros/noetic
-- This workspace overlays: /home/hb/carto_ws/cartographer_detailed_comments_ws-master/install_isolated;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
[... 629 lines cut ...]
   13 | #error incompatible with your Protocol Buffer headers.  Please update
      |  ^~~~~
/usr/include/gazebo-11/gazebo/msgs/pose.pb.h:14:2: error: #error your headers.
   14 | #error your headers.
      |  ^~~~~
In file included from /usr/include/gazebo-11/gazebo/msgs/pose.pb.h:33,
                 from /usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:34,
                 from /usr/include/gazebo-11/gazebo/physics/PhysicsTypes.hh:26,
                 from /usr/include/gazebo-11/gazebo/common/Plugin.hh:41,
                 from /home/hb/Projects_ICT/ros_motion_planning/src/plugins/gazebo_plugins/pedestrian_sfm_plugin/include/pedestrian_sfm_plugin.h:30,
                 from /home/hb/Projects_ICT/ros_motion_planning/src/plugins/gazebo_plugins/pedestrian_sfm_plugin/src/pedestrian_sfm_plugin.cpp:21:
/usr/include/gazebo-11/gazebo/msgs/vector3d.pb.h:12:2: error: #error This file was generated by a newer version of protoc which is
   12 | #error This file was generated by a newer version of protoc which is
      |  ^~~~~
/usr/include/gazebo-11/gazebo/msgs/vector3d.pb.h:13:2: error: #error incompatible with your Protocol Buffer headers. Please update
   13 | #error incompatible with your Protocol Buffer headers.  Please update
      |  ^~~~~
/usr/include/gazebo-11/gazebo/msgs/vector3d.pb.h:14:2: error: #error your headers.
   14 | #error your headers.
      |  ^~~~~
In file included from /usr/include/gazebo-11/gazebo/msgs/pose.pb.h:34,
                 from /usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:34,
                 from /usr/include/gazebo-11/gazebo/physics/PhysicsTypes.hh:26,
                 from /usr/include/gazebo-11/gazebo/common/Plugin.hh:41,
                 from

Xiawenlong-bug, 2024-08-05:
同样遇到这个问题。

ai-winter, 2024-08-19:
@Xiawenlong-bug @CHB-learner 

My protoc version is 3.6.1

closed 2024-09-22 by ai-winter