ai-winter/ros_motion_planning #90: 雷达数据和坐标系转换

t1tqq, 2024-07-20:
这个问题怎么解决
[ WARN] [1721482204.267190101, 30.213000000]: No laser scan received (and thus no pose updates have been published) for 30.213000 seconds.  Verify that data is being published on the /scan topic.
[ WARN] [1721482204.618012560, 30.563000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.

ai-winter, 2024-08-04:
@t1tqq 

Please check the tf-tree and ensure that AMCL operates normally.

Feijiake, 2024-08-06:
我也有着相同的问题,
`(base) qing@Qing:~$ rosnode list
/amcl
/gazebo
/gazebo_gui
/map_server
/move_base
/robot_state_publisher
/rosout
/rviz
/tf_echo_1722959802789771374
(base) qing@Qing:~$ rostopic echo /scan
WARNING: no messages received and simulated time is active.
Is /clock being published?

`

> @t1tqq
> 
> Please check the tf-tree and ensure that AMCL operates normally.

ZhanyuGuo, 2024-08-23:
@t1tqq @Feijiake Hi! Can you provide more information such as commands, configuration (i.e., user_config.yaml), and we'll try to reproduce the issue.

YsqRos, 2024-11-23:
同样的问题

cwgxy, 2025-05-15:
face the same issue, I just follow the readme till ./main.sh.   nerver change use_config.yaml

cwgxy, 2025-05-15:
map: "warehouse"
world: "warehouse"
rviz_file: "sim_env.rviz"

robots_config:
  - robot1_type: "turtlebot3_waffle"
    robot1_global_planner: "hybrid_astar"
    robot1_local_planner: "rpp"
    robot1_x_pos: "0.0"
    robot1_y_pos: "0.0"
    robot1_z_pos: "0.0"
    robot1_yaw: "0.0"

# * global planner
#   * graph_planner
#     * astar
#     * jps
#     * gbfs
#     * dijkstra
#     * d_star
#     * lpa_star
#     * voronoi
#     * d_star_lite
#     * theta_star
#     * lazy_theta_star
#     * hybrid_astar
#
#   * sample_planner
#     * rrt
#     * rrt_star
#     * rrt_connect
#     * informed_rrt
#     * quick_informed_rrt
#
#   * evolutionary_planner
#     * aco
#     * pso
#     * ga
#   * lazy_planner
#
# * local planner
#   * dwa
#   * pid
#   * apf
#   * rpp
#   * lqr
#   * static

plugins:
  # pedestrians: "pedestrian_config.yaml"
  # obstacles: "obstacles_config.yaml"
  map_layers: "map_layers_config.yaml"

YsqRos, 2025-05-15:
您发给杨舜清的邮件已经收到，谢谢。