ai-winter/ros_motion_planning #94: roslaunch运行main.launch失败

CharlotteDucky, 2024-09-21:
作者你好，我在编译完成之后，使用roslaunch sim_env main.launch启动main.launch报错，RLException: Roslaunch got a 'No such file or directory' error while attempting to run:

/home/wanjie/ros_motion_planning/src/plugins/dynamic_xml_config/plugins/obstacles_generate_ros.py user_config.yaml __name:=obstacles_generate __log:=/home/wanjie/.ros/log/a99a6eba-77f2-11ef-9a10-8bd198d20b00/obstacles_generate-2.log

Please make sure that all the executables in this command exist and have
executable permission. This is often caused by a bad launch-prefix.
The traceback for the exception was written to the log file

在那之前我已经source devel/setup.bash，并将工作空间路径加入.bashrc文件，并且直接使用的你的工作空间作为根目录。实在没找到解决办法，请帮忙看一下如何解决，感谢！！！
[image]

CharlotteDucky, 2024-09-21:
与此同时，运行main.sh也是报一样的错误无法显示仿真的

CharlotteDucky, 2024-09-21:
我已经解决了，将obstacles_generate_ros.py的声明改成python3

ai-winter, 2024-09-22:
@CharlotteDucky 

ok. Please run the system in Ubuntu20.

closed 2024-09-22 by ai-winter