ai-winter/ros_motion_planning #100: APF支持动态避障吗？

caiweijie2000, 2024-12-06:
您好，首先很感谢项目的开源，对我的帮助很大！我现在想用RRT和APF实现在动态环境中的融合路径规划，我把move_base_param.yaml中的planner_frequency设为 0.01，使得只在开始时进行一次全局规划，启用obstacles_config.yaml，使得地图中出现全局地图中未记录的障碍物，但是实际运行中发现并不可以动态避障，发现在APF中似乎只用到了引力，并未用到斥力，因为斥力始终算的是world坐标系中(0,0)处的斥力，而不是机器人当前位置的斥力
[image]
我改为机器人实时位置的斥力后，移动机器人时一直显示
[image]
[image]
请问这是什么原因，有什么办法可以利用APF实现动态避障吗？
期待您的回复

omigeft, 2024-12-06:
[image]
The code you presented is different from the latest version of this repository. We recommend that you update the code version.
您展示的代码与仓库中最新版本的不同。我们推荐您更新代码版本。

MasterYC, 2024-12-10:
> ![image](https://private-user-images.githubusercontent.com/131282526/393156938-1e94e728-293c-405a-8d62-177b5f39452c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

最新版其实仍然有问题。current_ps是map坐标系下的坐标，但是却直接调用了local_map的worldToMap，local_map的主体坐标系明明是odom，我不知道为什么作者的所有local_planner的代码都是这样，每次都要转换到map坐标系下来计算