ai-winter/ros_motion_planning #101: catkin_make error when I start to "./build.sh"

KetenBieber, 2024-12-12:
[image]

This seems to have some grammar errors, but I don't think there should be this problem. I have already executed the previous conan environment in order, but at this step, this problem has arisen and has not been resolved yet! Does it mean there are requirements for the C++version? I am using C++17, and I tried to explicitly specify C++17 after the instruction in catkin_make, but it was to no avail; (That is to say, - DCMAKE_CXX-STANDARD=17 was added after catkin_make in the build.sh script), but this cannot be resolved

TheFirstBoy1, 2024-12-22:
Did you resolve it?I have the same problem.Can you share some experience.

KesiroH, 2025-01-01:
It seems that we have the same problem...

closed 2025-01-03 by ai-winter