ai-winter/ros_motion_planning #104: build.sh error

chan-yuu, 2024-12-25:
/usr/bin/ld: cannot find -losqp
/usr/bin/ld: cannot find -lunwind-ptrace
/usr/bin/ld: cannot find -lunwind-setjmp
/usr/bin/ld: cannot find -lunwind-coredump
/usr/bin/ld: cannot find -lunwind-generic
/usr/bin/ld: cannot find -lunwind
collect2: error: ld returned 1 exit status
make[2]: *** [core/controller/controller/CMakeFiles/controller.dir/build.make:122: /home/cyun/learn_ws/11/ros_motion_planning/devel/lib/libcontroller.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:7852: core/controller/controller/CMakeFiles/controller.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

chan-yuu, 2024-12-25:
1.install libunwind：
```
sudo apt install libunwind-setjmp*
sudo apt-get install libunwind-dev
```
2.install osqp:
```
git clone --recursive https://github.com/oxfordcontrol/osqp.git
cd osqp

```
```
mkdir build
cd build
cmake ..
make
sudo make install
```

`sudo ldconfig`

Re execute build.sh：
[image]

ZhanyuGuo, 2024-12-26:
Thanks for your solution!

closed 2024-12-26 by ZhanyuGuo

liusumengshang, 2024-12-26:
[image]
请问这个问题怎么去解决

chan-yuu, 2024-12-26:
`pip install rospkg`